openrazer/openrazer #1454: Polychromatic doesn't open

BruceJi7, 2021-03-10:
### Distribution

Ubuntu 20.04.2 LTS, 64-bit
GNOME 3.36.8
OpenRazer 2.9.0~ubuntu20.04.1

### Support

- [/ ] Have you tried rebooting?
- [ /] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

If I run the controller or tray applet, nothing happens.
If I run the command `polychromatic-controller` I get the following error:

```
Traceback (most recent call last):
  File "/usr/bin/polychromatic-controller", line 26, in <module>
    import distro
ModuleNotFoundError: No module named 'distro'
```
Not related, but apparently I can't install the module 'distro', pip, pip3 and python3 -m pip all fail :S

I followed the troubleshooting procedure and can confirm the daemon works.

lah7, 2021-03-10:
Please install the `python3-distro` package:

    sudo apt install python3-distro

(The application never uses `pip`)

Regarding the invisible tray applet, this is because GNOME (as used in Ubuntu) considers the tray "deprecated" and you'll need to install `gnome-shell-extension-appindicator` as [per the docs](https://polychromatic.app/docs/distros/#tray-applet-vs-desktop-environments).

    sudo apt install gnome-shell-extension-appindicator

Depending on what hardware you have, you may find the development version in `ppa:polychromatic/edge` offers better support in the interim then this (presumably) v0.3.12 build.

For any future issues of the app, [please raise them in Polychromatic's repository](https://github.com/polychromatic/polychromatic/issues) - it's no fault of OpenRazer 🙂

BruceJi7, 2021-03-12:
Ah! I thought I was in the Polychromatic's repository, my bad.
I ran that command, but I get exactly the same message. Maybe my Python path is messed up?

BruceJi7, 2021-03-12:
[image]

BruceJi7, 2021-03-12:
I'll raise this issue over on the right repo, anyway

BruceJi7, 2021-03-12:
My dude, I followed these instructions - https://github.com/polychromatic/polychromatic/issues/293#issuecomment-712781234 - and it opens fine :D Thanks!

z3ntu, 2021-03-12:
Closing then.

closed 2021-03-12 by z3ntu