openrazer/openrazer #1455: Support for Razer Huntsman Mini

ne04, 2021-03-11:
### Device Info

* Name: Razer Huntsman Mini
* Product Number: RZ03-03390500-R3U1
* Model Number: PM2103F45507645

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

Bus 003 Device 002: ID 1532:0257 Razer USA, Ltd Razer Huntsman Mini
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0257 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     177
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              2 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 240 Mar 10 18:34 .
drwxr-xr-x 4 root root 620 Mar 10 18:34 ..
lrwxrwxrwx 1 root root  10 Mar 10 18:34 usb-Logitech_Advanced_Corded_Mouse_M500s_206138665342-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 Mar 10 18:34 usb-Logitech_Advanced_Corded_Mouse_M500s_206138665342-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 Mar 10 18:34 usb-Logitech_Advanced_Corded_Mouse_M500s_206138665342-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root   9 Mar 10 18:34 usb-Logitech_Advanced_Corded_Mouse_M500s_206138665342-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Mar 10 18:34 usb-Razer_Razer_Huntsman_Mini_00000000001A-event-if01 -> ../event5
lrwxrwxrwx 1 root root   9 Mar 10 18:34 usb-Razer_Razer_Huntsman_Mini_00000000001A-event-if03 -> ../event8
lrwxrwxrwx 1 root root   9 Mar 10 18:34 usb-Razer_Razer_Huntsman_Mini_00000000001A-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Mar 10 18:34 usb-Razer_Razer_Huntsman_Mini_00000000001A-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Mar 10 18:34 usb-Razer_Razer_Huntsman_Mini_00000000001A-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Mar 10 18:34 usb-Razer_Razer_Huntsman_Mini_00000000001A-if02-mouse -> ../mouse0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

Don't understand how to do packet captures but if it's needed I can get those.

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Cant get it to run on my machine sorry

danielmatthewsgrout, 2021-03-16:
Same for me - it doesn't recognise it.

From dmesg:
```[  924.983929] usb 1-4.2: new full-speed USB device number 14 using xhci_hcd
[  925.095027] usb 1-4.2: New USB device found, idVendor=1532, idProduct=0257, bcdDevice= 2.00
[  925.095028] usb 1-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  925.095029] usb 1-4.2: Product: Razer Huntsman Mini
[  925.095030] usb 1-4.2: Manufacturer: Razer
[  925.095030] usb 1-4.2: SerialNumber: 00000000001A
[  925.111038] input: Razer Razer Huntsman Mini as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.2/1-4.2:1.0/0003:1532:0257.0018/input/input51
[  925.172503] hid-generic 0003:1532:0257.0018: input,hidraw7: USB HID v1.11 Keyboard [Razer Razer Huntsman Mini] on usb-0000:00:14.0-4.2/input0
[  925.174547] input: Razer Razer Huntsman Mini Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.2/1-4.2:1.1/0003:1532:0257.0019/input/input52
[  925.232805] input: Razer Razer Huntsman Mini Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.2/1-4.2:1.1/0003:1532:0257.0019/input/input53
[  925.232875] input: Razer Razer Huntsman Mini System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.2/1-4.2:1.1/0003:1532:0257.0019/input/input54
[  925.232907] input: Razer Razer Huntsman Mini as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.2/1-4.2:1.1/0003:1532:0257.0019/input/input55
[  925.232972] hid-generic 0003:1532:0257.0019: input,hidraw8: USB HID v1.11 Keyboard [Razer Razer Huntsman Mini] on usb-0000:00:14.0-4.2/input1
[  925.234306] input: Razer Razer Huntsman Mini as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.2/1-4.2:1.2/0003:1532:0257.001A/input/input56
[  925.234366] hid-generic 0003:1532:0257.001A: input,hidraw9: USB HID v1.11 Mouse [Razer Razer Huntsman Mini] on usb-0000:00:14.0-4.2/input2
[  925.237484] input: Razer Razer Huntsman Mini Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.2/1-4.2:1.3/0003:1532:0257.001B/input/input57
[  925.296690] hid-generic 0003:1532:0257.001B: input,hiddev2,hidraw10: USB HID v1.11 Device [Razer Razer Huntsman Mini] on usb-0000:00:14.0-4.2/input3

```

sebw, 2021-03-19:
Duplicate #1230

z3ntu, 2021-03-21:
Yes, duplicate of #1230

closed 2021-03-21 by z3ntu