openrazer/openrazer #1457: modprobe razerkbd : Exec format error

vlad34, 2021-03-13:
### Distribution

Fedora 33, KDE

Kernel: 5.10.21-200.fc33.x86_64

### Support

- [ ] Have you tried rebooting?

Yes.

- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I can't load module "razerkbd" anymore, do you have any idea?

`$ sudo dkms install --verbose openrazer-driver/2.9.0
Module openrazer-driver/2.9.0 already installed on kernel 5.10.21-200.fc33.x86_64/x86_64
`

`$ sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Exec format error`

`$ sudo modprobe razerkbd; dmesg | tail
modprobe: ERROR: could not insert 'razerkbd': Exec format error
[35438.505702] razermouse: version magic '5.10.20-200.fc33.x86_64 SMP mod_unload ' should be '5.10.21-200.fc33.x86_64 SMP mod_unload '
[35464.047923] usb 1-1: new low-speed USB device number 8 using xhci_hcd
[35464.185528] usb 1-1: New USB device found, idVendor=05a4, idProduct=9840, bcdDevice= 1.10
[35464.185531] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[35464.185532] usb 1-1: Product: USB Compliant Keypad
[35464.192615] input: USB Compliant Keypad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:05A4:9840.000F/input/input54
[35464.245302] hid-generic 0003:05A4:9840.000F: input,hidraw3: USB HID v1.10 Keyboard [USB Compliant Keypad] on usb-0000:00:14.0-1/input0
[41482.297952] razerkbd: version magic '5.10.20-200.fc33.x86_64 SMP mod_unload ' should be '5.10.21-200.fc33.x86_64 SMP mod_unload '
[42148.136432] razerkbd: version magic '5.10.20-200.fc33.x86_64 SMP mod_unload ' should be '5.10.21-200.fc33.x86_64 SMP mod_unload '
[42166.171251] razerkbd: version magic '5.10.20-200.fc33.x86_64 SMP mod_unload ' should be '5.10.21-200.fc33.x86_64 SMP mod_unload '`

Thanks for the help.

z3ntu, 2021-03-13:
What's the output of `uname -r` ? You seem to be running the 5.10.20 version while 5.10.21 is on your file system (and what dkms built against)

vlad34, 2021-03-13:
The result,

`$ uname -r
5.10.21-200.fc33.x86_64`

z3ntu, 2021-03-13:
Try this:
```
sudo dkms remove openrazer-driver/2.9.0 --all
sudo dkms install openrazer-driver/2.9.0
```

vlad34, 2021-03-13:
Result:

https://pastebin.com/wRC1xX35

Strange message during uninstallation and installation ...

`depmod: ERROR: fstatat(4, razerkraken.ko.xz): No such file or directory`
`depmod: ERROR: fstatat(4, razeraccessory.ko.xz): No such file or directory`
`depmod: ERROR: fstatat(4, razerkbd.ko.xz): No such file or directory`
`depmod: ERROR: fstatat(4, razermouse.ko.xz): No such file or directory`

vlad34, 2021-03-13:
I reboot,

It's good,I had to reboot computer after reinstalling OpenRazer

Thanks,

closed 2021-03-13 by vlad34