openrazer/openrazer #1458: Support for Razer Huntsman V2 Analog

joeknock90, 2021-03-13:
### Device Info

* Name: Razer Huntsman V2 Analog
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
```
Bus 001 Device 003: ID 1532:0266 Razer USA, Ltd Razer Huntsman V2 Analog
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0266 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Huntsman V2 Analog
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     212
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0018  1x 24 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        4
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              2 Razer Huntsman V2 Analog
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```
```
lrwxrwxrwx 1 root root  10 Mar 13 15:02 usb-Razer_Razer_Huntsman_V2_Analog-event-if01 -> ../event17
lrwxrwxrwx 1 root root  10 Mar 13 15:02 usb-Razer_Razer_Huntsman_V2_Analog-event-if04 -> ../event25
lrwxrwxrwx 1 root root  10 Mar 13 15:02 usb-Razer_Razer_Huntsman_V2_Analog-event-kbd -> ../event20
lrwxrwxrwx 1 root root  10 Mar 13 15:02 usb-Razer_Razer_Huntsman_V2_Analog-if01-event-kbd -> ../event21
lrwxrwxrwx 1 root root  10 Mar 13 15:02 usb-Razer_Razer_Huntsman_V2_Analog-if02-event-mouse -> ../event24
lrwxrwxrwx 1 root root   9 Mar 13 15:02 usb-Razer_Razer_Huntsman_V2_Analog-if02-mouse -> ../mouse1
```
### Packet Captures
[Razer Huntsman v2 Packet Cap.tar.gz](https://github.com/openrazer/openrazer/files/6135506/Razer.Huntsman.v2.Packet.Cap.tar.gz)

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
Attached
### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Incoming via USB passthrough to a VM, will update post.

joeknock90, 2021-03-14:
Took a packet cap with USB pass through to windows. Started with synapse closed and then opened synapse. There's a LOT of chatter. 

File is attached. I'm going to try to figure out how to filter out the regular chatter and then take packet captures of changing key settings, especially of the "Hypershift" mode and game controller modes. 

This one might be a beast to reverse engineer. Hopefully with enough packet captures I can start on really reverse engineering the protocol, but it's been a while since I've worked with wireshark. 

[Razer Huntsman v2 USB Cap - Synapse Open.tar.gz](https://github.com/openrazer/openrazer/files/6136692/Razer.Huntsman.v2.USB.Cap.-.Synapse.Open.tar.gz)

z3ntu, 2021-03-14:
Thanks for the pcap, here are some pointers for the implementation if you want to try:

Use `razer_chroma_extended_matrix_set_custom_frame` for custom frame, `razer_chroma_extended_matrix_effect_*` functions for effects (and ZERO_LED led id), `razer_chroma_extended_matrix_brightness` for brightness (also ZERO_LED),  transaction id 0x1f for everything

And use `MATRIX_DIMS = [8, 23]` for the daemon

joeknock90, 2021-03-14:
Great info. Thanks! I'll take some more pcaps. I'm hoping I can start digging into the analog settings as well, but I might just have to set those in Windows as hardware profiles for now.

z3ntu, 2021-03-14:
I don't think more pcaps won't help much, I'd recommend you just start with the implementation (see recent device additions in the git log) - at least that should get you lighting working.
If Synapse indeed sends some more commands to the firmware for the analog parts (I do see unknown messages in the pcap), then you should try to extract them by starting the capture, doing a single action there (and document what was done), and stop the capture again, but I'd delay this until basic support is working.

joeknock90, 2021-03-14:
Good call. Will start on that first. I am not a developer, Linux admin by trade, but I will start seeing if I can get something working initially.

z3ntu, 2021-03-14:
Do you have any specific questions I could answer?

joeknock90, 2021-03-15:
I think I'm actually starting to understand here just by looking through previous pull requests. I'll definitely have some more specific questions soon.

joeknock90, 2021-03-15:
Yeah. This is way over my head. I can't get most of the changes I make to work. Just keep getting errors running the daemon. 

Is there any development documentation for the project?

WrathofthePast, 2021-04-29:
> Ok, so I've got the daemon running with the device added. How do I get polychromatic/razergenie to see it now. Because they still don't seem to think it' exists. UDEV rules are set and I believe it's being picked up and bound to the razerkbd driver, though I'm not entirely sure.

Are you still stuck on this?

J5cott, 2021-06-19:
I would like to purchase the Huntsman V2. I use Pop OS and would love to donate in order to get linux support for the device.  I'll donate 20 bucks now and 80 bucks when this issue is resolved.

WrathofthePast, 2021-06-19:
@J5Scott I already own one & would love to develop support for it, but I don't have experience with this type of coding. I also don't have a Linux machine to test this on besides my Ubuntu virtual machine (not useful). What I can offer: if you fund this, I'll fund it on 1kc/razer-macos#250, so macOS support can be added. They port the drivers from here, so once Linux is supported, MacOS can be.

J5cott, 2021-06-19:
@WrathofthePast I'm not sure how to 'fund' or 'sponsor' an issue.  Is there a feature on Github that I should use to do this?

WrathofthePast, 2021-06-22:
@J5cott If you go to the "< > Code" tab on openrazer/openrazer, you'll see three PayPal links to sponsor it under "Sponsor this project" on the right side of the page. Note that these links are to the individuals in charge of openrazer & there's no "funded"/"sponsored" label for issues either. However, there is a "help wanted" label & you'll be able to put a note on a PayPal transfer saying what the money is for. They'll put that label on this issue or take up it themselves.

J5cott, 2021-06-22:
Ok @WrathofthePast I sent @terrycain another 40 bucks and included the url to this issue in the notes on the payment.

terrycain, 2021-06-23:
I dont think we ever decided on sponsoring issues specifically, mainly the paypal links were for donations.

That being said, I now feel obliged to work on this... after a 4 year absence :smile: 

Who has a Huntsman V2 Analog ? Also what does razer list as the differences between the analog and the huntsman elite? As I would not be surprised if they're fundamentally the same keyboard.

jscottPTC, 2021-06-23:
@terrycain it sounds like @WrathofthePast currently owns one:

> @j5scott I already own one & would love to develop support for it, but I don't have experience with this type of coding. I also don't have a Linux machine to test this on besides my Ubuntu virtual [...]

I'm not sure if he is willing to share it though....

WrathofthePast, 2021-06-23:
@terrycain @jscottPTC I'll help however I can. I have two Huntsman v2 - Analogs since Amazon lost the original & it still showed up. I left that one unopened. The OP posted the pcap earlier in the thread, so that step is already done; however, let me know if a new pcap is needed for any reason.

WrathofthePast, 2021-06-27:
@terrycain  Here's the differences Razer lists for the Huntsman V2 Analog & Huntsman Elite (Clicky Optical & Linear Optical):

1. Switch Type: "Analog Optical Switch" vs. "Clicky Optical" & "Linear Optical"
2. Connectivity: USB-C vs USB-A
3. Lighting: "Razer Chroma RGB customizable backlighting with 16.8 million color options" vs "Razer Chroma backlighting with underglow"
4. Onboard Memory: "Hybrid onboard storage - up to 5 **keybinding** profiles" vs "Hybrid onboard storage - up to 5 profiles"
5. Others:
- "N-key roll-over with anti-ghosting" vs. "10 key roll-over with anti-ghosting"
6. Passthrough: USB-A 3.0 Passthrough vs None

Just to include the answers to both of your questions in the same comment:
> @terrycain @jscottPTC I'll help however I can. I have two Huntsman v2 - Analogs since Amazon lost the original & it still showed up. I left that one unopened. The OP posted the pcap earlier in the thread, so that step is already done; however, let me know if a new pcap is needed for any reason.

WrathofthePast, 2021-06-27:
>2. Switch Type: "Analog Optical Switch" vs. "Clicky Optical Switch" & "Linear Optical Switch"

The analog switches allow each and every key to have two functions depending how far they are pressed down & that activation point for both can adjusted.

mhardeman, 2021-07-03:
If there's a project dev willing to make a public statement of commitment on this issue to an honest effort to add support for the Hubatsman V2 Analog, I am willing to immediately fund an amount equal to the acquisition price of the keyboard and an extra $50 for their interest.

z3ntu, 2021-07-05:
@mhardeman What are you looking for exactly? Support in OpenRazer like all other devices, meaning lighting + special buttons on the keyboard working? I'm not familiar if the analog switches need any special (kernel) support or what they're used for exactly.

mhardeman, 2021-07-05:
@z3ntu My primary interest is lighting.  Indeed, control of the lighting would fully satisfy me with respect to the $300 offer I've made.  The analog switches are every key on the keyboard.  It's an analog light-beam measurement system that allows you to control the precise key movement for key activation and separately the travel distance for release of the key.  Additionally, you can specify two different travel distances for keys allowing them to act as two different keys.  That part isn't really important to me, but for what it's worth, I believe the implementation is entirely in the keyboard firmware.  My belief is that it's just a matter of configuring it from the USB side.

z3ntu, 2021-07-05:
@mhardeman Sounds good. The keyboard is €269.99 in Europe (https://www.razer.com/eu-en/gaming-keyboards/Razer-Huntsman-V2-Analog/RZ03-03610100-R3M1), so converted with Google it's ~320 USD (bad time for converting USD to EUR :smile:). I think buying in the US and shipping to Europe would still be more expensive though due to shipping itself and import tax (the latter itself is probably more than 60€).

mhardeman, 2021-07-05:
@z3ntu Hello, I understand there can be differences in exchange rate as well as pricing for electronics in Europe.  My offer was intended as the full cost of acquisition of one Huntsman V2 Analog keyboard as well as approximately $50.  I have sent EUR 320.00 via PayPal, and I have specified that PayPal fees should be paid from my side.  My understanding is that this gets you the full EUR 320.  Is that acceptable?

z3ntu, 2021-07-05:
If you send it as "for family & friends" then it should be a free transfer I think (apart from the exchange rate from your side I guess). If you send it "as payment for something" (or whatever it's called) then PayPal definitely takes a big chunk, so a recent 5€ donation became 4.48€... But yes I'd order it asap then and start the implementation when I have it :)

z3ntu, 2021-07-05:
@mhardeman Cheers, I'll keep you updated!

mhardeman, 2021-07-05:
@z3ntu Thanks much!  It's a pleasure to contribute to a great project that's supporting this hardware on other OS's.  Enjoy the keyboard.  I just got mine and I'm loving it!  The loud cracking noise when I slam the enter key could wake the dead!

z3ntu, 2021-07-11:
PR with basic support is here: #1554

Proper analog key support is non-trivial as I'm not aware of any other analog keyboards having kernel-level support with matching user space to handle the different way of input.

mhardeman, 2021-07-11:
Thanks for jumping on this so quickly!  I'll be watching out for it to get merged to master.  I'll test it on my linux system and if all goes well, I'll either try to get the people to pull it into the macOS port or do that part myself.

WrathofthePast, 2021-07-14:
@mhardeman Are you alluding to porting this to 1kc/razer-macos#250 or FruityRazer/FruitUI? I was going to fund the issue on razer-macos, so if you're going to do it there, I can send you the money directly.

mhardeman, 2021-07-14:
@WrathofthePast It was my intent to ultimately wait for this work to make it into the main branch of this project, and then sponsor work on razer-macos to support this.

Having said that, I recently tried this and had trouble getting it to take on my keyboard.  I did make a few mods to the patch on this issue and was able to set a static color, which was my goal.  At the end of the day, I wanted a simple binary that I could put in my user login profile that would set my keyboard purple.  I've achieved that in a hacky way now, but would still like to see this thing through to completion.

closed 2021-08-15 by z3ntu

J5cott, 2021-08-15:
@z3ntu since it seems you have done most of the work to resolve this issue I would like to donate to you.  I don't see where I can do that on your user.  What do you recommend?

z3ntu, 2021-08-16:
@J5cott There's a link to my PayPal under "Sponsor this project" in the main view of the repo (https://github.com/openrazer/openrazer)

PR #1554 (2021-08-15): Add support for Razer Huntsman V2 Analog
Without driver mode support for now, because that needs special handling
for even the normal key input to work (because of analog keys).

Fixes #1458

---

How to build and test: https://github.com/openrazer/openrazer/wiki/Building-a-package

CC @mhardeman