openrazer/openrazer #1459: Fn keys are not working on 1532:0241 BlackWidow (2019)

fjpanag, 2021-03-17:
### Distribution

Fedora 33

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The Fn keys are not working.  
I press the Fn, the F keys turn white, but pressing them does absolutely nothing.

This is only happening when the openrazer daemon is running.  
If it is stopped, the Fn keys functionality returns to normal.

z3ntu, 2021-03-17:
Try building the package from latest git master, this should be fixed there.

https://github.com/openrazer/openrazer/wiki/Building-a-package#fedora

fjpanag, 2021-03-17:
@z3ntu I tried building the latest commit in master.  
It still does not work.

z3ntu, 2021-03-22:
Let's track this in #1466, and read the latest comment if you want to help.

closed 2021-03-22 by z3ntu