openrazer/openrazer #1460: Support for Razer Base Station V2 Chroma

harrisonthorne, 2021-03-19:
### Device Info

* Name: Razer Base Station V2 Chroma
* Product Number: RC21-01510100
* Model Number: RC21-0151

### Device Information

```
Bus 001 Device 005: ID 1532:0f20 Razer USA, Ltd Razer Base Station V2 Chroma
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f20 
  bcdDevice            2.00
  iManufacturer           1 RAZER
  iProduct                2 Razer Base Station V2 Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               2.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
total 0
drwxr-xr-x 2 root root 280 Mar 19 13:36 .
drwxr-xr-x 4 root root 700 Mar 19 13:36 ..
lrwxrwxrwx 1 root root   9 Mar 19 13:36 usb-Generic_Razer_Base_Station_V2_Chroma-event-if03 -> ../event6
lrwxrwxrwx 1 root root   9 Mar 19 13:36 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Mar 19 13:36 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Mar 19 04:15 usb-MSI_PRO_CARBON_A02018070613-event-if00 -> ../event5
lrwxrwxrwx 1 root root  10 Mar 19 04:15 usb-RAZER_Razer_Base_Station_V2_Chroma-event-if01 -> ../event10
lrwxrwxrwx 1 root root   9 Mar 19 04:15 usb-RAZER_Razer_Base_Station_V2_Chroma-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Mar 19 04:15 usb-RAZER_Razer_Base_Station_V2_Chroma-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root   9 Mar 19 04:15 usb-RAZER_Razer_Base_Station_V2_Chroma-mouse -> ../mouse2
lrwxrwxrwx 1 root root   9 Mar 19 04:15 usb-Razer_Razer_Cynosa_Chroma-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Mar 19 04:15 usb-Razer_Razer_Cynosa_Chroma-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Mar 19 04:15 usb-Razer_Razer_Cynosa_Chroma-if02-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Mar 19 04:15 usb-Razer_Razer_Cynosa_Chroma-if02-mouse -> ../mouse0
```
### TODO:

### Packet Captures

### Synapse Screenshots

nightsky30, 2021-03-20:
Support for this device was already added to the master branch via #1403

harrisonthorne, 2021-03-20:
Oh! Cool! Thanks :)

closed 2021-03-20 by harrisonthorne