openrazer/openrazer #1463: Support for Naga X 2021

BuyMyMojo, 2021-03-22:
### Device Info

* Name: Razer Naga X
* Product Number: <See under device>
* Model Number: RZ01-03590100-R3U1

### Device Information
```
[buymymojo@buymymojo ~]$ lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:0096 Razer USA, Ltd Razer Naga X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0096 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga X
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
[buymymojo@buymymojo ~]$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 220 Mar 22 12:24 .
drwxr-xr-x 6 root root 680 Mar 22 12:24 ..
lrwxrwxrwx 1 root root   9 Mar 22 12:23 ckb-CORSAIR_K68_RGB_Mechanical_Gaming_Keyboard_vKB_-event -> ../event2
lrwxrwxrwx 1 root root   9 Mar 22 12:23 ckb-CORSAIR_K68_RGB_Mechanical_Gaming_Keyboard_vM_-event -> ../event3
lrwxrwxrwx 1 root root   9 Mar 22 12:24 usb-046d_C922_Pro_Stream_Webcam_F5E9FC1F-event-if00 -> ../event4
lrwxrwxrwx 1 root root  10 Mar 22 12:22 usb-Blue_Microphones_Yeti_Stereo_Microphone_797_2018_10_16_83371-event-if03 -> ../event13
lrwxrwxrwx 1 root root   9 Mar 22 12:22 usb-Razer_Razer_Naga_X-event-if01 -> ../event9
lrwxrwxrwx 1 root root   9 Mar 22 12:22 usb-Razer_Razer_Naga_X-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Mar 22 12:22 usb-Razer_Razer_Naga_X-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Mar 22 12:22 usb-Razer_Razer_Naga_X-if02-event-kbd -> ../event12
lrwxrwxrwx 1 root root   9 Mar 22 12:22 usb-Razer_Razer_Naga_X-mouse -> ../mouse1

```

LivingWithHippos, 2021-03-31:
If I fire up the windows VM I can change/create profiles and it works on linux even after I close the vm, so that's pretty good already.

Here is a capture for the buttons, I cycled from 1 to 12 on the side keyboard, left, right click, mouse wheel click, dpi button click and then moved the wheel. Lightining was turned off

[razer_naga_x.pcap.zip](https://github.com/openrazer/openrazer/files/6237588/razer_naga_x.pcap.zip)

z3ntu, 2021-03-31:
These pcaps only contain one DPI and one DPI stages call - at least the only ones in a known format.

Myrnedraith, 2021-05-12:
Any word on this/anything I can do to help as a Naga X owner to get this working?

BuyMyMojo, 2021-05-18:
> Any word on this/anything I can do to help as a Naga X owner to get this working?

I cant currently do this but if you can record the usb info while changing settings in the Razer software then that'd be great. like changing the buttons on the side, the colours, dpi and hypershift maybe?
https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

BuyMyMojo, 2021-06-28:
I have no idea if this helps but I tried to record my USB info using [USBPcap](https://desowin.org/usbpcap/) while actually booted into windows

[RazerNagaX-USBPcap.zip](https://github.com/openrazer/openrazer/files/6723591/RazerNagaX-USBPcap.zip)

z3ntu, 2021-06-30:
@BuyMyMojo Thanks. There's what's in the requests (for somebody looking to implement and test it on the device):
* for all requests transaction id 0x1f
* `razer_chroma_extended_matrix_brightness` with ZERO_LED
* `razer_chroma_extended_matrix_effect_*` effects with ZERO_LED
* `razer_chroma_extended_matrix_set_custom_frame` for custom frame, use `MATRIX_DIMS = [1, 2]` in daemon config
* dpi & poll rate normal; support for dpi stages

daenney, 2021-11-20:
There's #1677 now which should hopefully work. I'm still waiting on the actual hardware to get delivered, so that PR is currently marked as draft since it's based purely on what was said in this issue.

reoccurcat, 2021-12-01:
Any update on this? I installed the daily build and I still can't use my Naga X. I checked today. Polychromatic shows that openrazer hasn't registered it.

daenney, 2021-12-01:
There's a long discussion in the PR itself, #1677, about issues getting this to work. I'd suggest reading up on that and keeping discussions there.

nicolas3366, 2022-05-11:
Hi everyone. Is there anything I as a non programmer can contribute? I have a Naga X, I have a Mac and I have a PC.

archer31, 2022-05-11:
Sadly the person who was working on this abandoned their pull request as they ran out of time to work on it. The maintainers might decide to mark this issue to indicate that it is no longer being worked but is looking for a contributer.

Someone who is more experienced with this library might look to see if configuration from the Razer Naga Pro could be leveraged to support this device as it supports the same side buttons with a few extra options.

nicolas3366, 2022-05-12:
Ok, that is unfortunate. Thanks for the info!

bgrabow, 2022-06-29:
On the above-referenced PR, there's a comment suggesting that key rebinds on the Naga X persist from Windows to Linux when dual booting on the same machine.

https://github.com/openrazer/openrazer/pull/1677#issuecomment-1019542442

I confirmed that this works: I installed Synapse on my Windows installation, rebound keys, then switched to my Linux Mint installation and the rebinds were still intact. Additionally, the rebinds persist ACROSS machines. When I plug the mouse into my Macbook (separate machine, separate OS) the rebinds persist still. They must be stored inside the mouse itself.

bufothefrog, 2022-07-14:
> On the above-referenced PR, there's a comment suggesting that key rebinds on the Naga X persist from Windows to Linux when dual booting on the same machine.
> 
> [#1677 (comment)](https://github.com/openrazer/openrazer/pull/1677#issuecomment-1019542442)
> 
> I confirmed that this works: I installed Synapse on my Windows installation, rebound keys, then switched to my Linux Mint installation and the rebinds were still intact. Additionally, the rebinds persist ACROSS machines. When I plug the mouse into my Macbook (separate machine, separate OS) the rebinds persist still. They must be stored inside the mouse itself.

I did this too, but unfortunately only keybinds and DPI stage info are saved to the device, not lighting information / other types of bindings.

bufothefrog, 2022-07-14:
I'm putting a $20 USD bounty on this getting solved, payable via Zelle/PayPal, cheritable donation or XMR. Feel free to remove if this isn't allowed.

bourquenoud, 2022-08-07:
I've added support for the Naga X in #1878

DagothUrTheFalse, 2022-10-18:
I seem to be having only one small issue with the naga x support when trying it, can't seem to set logo light but i might be misremembering if it has one, or it might be a front end issue instead of one with openrazer

z3ntu, 2022-10-29:
Indeed. Fixed by #1878

closed 2022-10-29 by z3ntu