openrazer/openrazer #1464: ModuleNotFoundError after upgrading to 3.0.0

Yoppez, 2021-03-22:
### Distribution

OS: Arch Linux x86_64 
Kernel: 5.11.8-arch1-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I recently upgraded the following packages to 3.0.0 using the "yay" AUR helper:
```
openrazer-daemon 3.0.0-1
openrazer-driver-dkms 3.0.0-1
openrazer-meta 3.0.0-1
python-openrazer 3.0.0-1
```
but after that upgrade i had some problems.

When opening Polychromatic Controller it says "Daemon cannot be relaunched", and when i try to run `openrazer-daemon -Fv` it prints this:
```
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 11, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
ModuleNotFoundError: No module named 'openrazer_daemon'
```
Running the commands on the troubleshooting page it seems that the kernel module is correctly installed.

z3ntu, 2021-03-22:
`pacman -Ql openrazer-daemon | head -n20` and `/usr/bin/python3 --version` please

Yoppez, 2021-03-22:
Here it is:
```
$ pacman -Ql openrazer-daemon | head -n20
openrazer-daemon /usr/
openrazer-daemon /usr/bin/
openrazer-daemon /usr/bin/openrazer-daemon
openrazer-daemon /usr/lib/
openrazer-daemon /usr/lib/python3.8/
openrazer-daemon /usr/lib/python3.8/site-packages/
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon-3.0.0-py3.8.egg-info/
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon-3.0.0-py3.8.egg-info/PKG-INFO
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon-3.0.0-py3.8.egg-info/SOURCES.txt
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon-3.0.0-py3.8.egg-info/dependency_links.txt
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon-3.0.0-py3.8.egg-info/top_level.txt
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/__init__.py
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/__pycache__/
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/__pycache__/__init__.cpython-38.pyc
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/__pycache__/daemon.cpython-38.pyc
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/__pycache__/device.cpython-38.pyc
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/__pycache__/keyboard.cpython-38.pyc
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/daemon.py
openrazer-daemon /usr/lib/python3.8/site-packages/openrazer_daemon/dbus_services/
```
```
$ /usr/bin/python3 --version
Python 3.9.2
```

z3ntu, 2021-03-22:
Somehow you seem to have built openrazer-daemon against Python 3.8 but you have Python 3.9 installed; not sure how you did that. Please try rebuilding openrazer-daemon and python-openrazer, so it's built against Python 3.9

Yoppez, 2021-03-22:
Ok, I solved it.

openrazer was built against Python 3.8 because of Anaconda, indeed if run ```/usr/bin/env python3``` it appears Python 3.8.
After uninstalling Anaconda and rebuilding openrazer-daemon and python-openrazer everything worked. Thanks a lot!

closed 2021-03-22 by Yoppez

z3ntu, 2021-03-22:
I recommend building AUR packages in a clean chroot, e.g. with https://wiki.archlinux.org/index.php/DeveloperWiki:Building_in_a_clean_chroot or with aurutils. That way these problems don't occur.

Yoppez, 2021-03-22:
Got it, thanks again.