openrazer/openrazer #1465: Support for Razer Basilisk V2 and Razer BlackWidow V3

krpanko, 2021-03-22:
### Device Info

* Name: Razer Basilisk V2 and Razer BlackWidow V3
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Bus 009 Device 003: ID 1532:0085 Razer USA, Ltd Razer Basilisk V2
Bus 009 Device 002: ID 1532:024e Razer USA, Ltd Razer BlackWidow V3
```

```
usb-Razer_Razer_Basilisk_V2-event-if01 -> ../event12
usb-Razer_Razer_Basilisk_V2-event-mouse -> ../event10
usb-Razer_Razer_Basilisk_V2-if01-event-kbd -> ../event11
usb-Razer_Razer_Basilisk_V2-if02-event-kbd -> ../event15
usb-Razer_Razer_Basilisk_V2-mouse -> ../mouse1
usb-Razer_Razer_BlackWidow_V3-event-if01 -> ../event5
usb-Razer_Razer_BlackWidow_V3-event-if04 -> ../event9
usb-Razer_Razer_BlackWidow_V3-event-kbd -> ../event3
usb-Razer_Razer_BlackWidow_V3-if01-event-kbd -> ../event4
usb-Razer_Razer_BlackWidow_V3-if02-event-mouse -> ../event8
usb-Razer_Razer_BlackWidow_V3-if02-mouse -> ../mouse0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

krpanko, 2021-03-22:
Basic stuff and keyboard and mouse work. However I can seem to do advanced settings, as polychromatic-controller doesnt find any Razer device.
 
![image](https://user-images.githubusercontent.com/81155141/111987695-d177ed00-8b0f-11eb-845e-057c0f096313.png

z3ntu, 2021-03-22:
Basilisk V2 is already supported in v3.0.0 (released yesterday). The BlackWidow V3 is tracked in #1319

closed 2021-03-22 by z3ntu

krpanko, 2021-03-22:
ok thanks.

much appreciated. Great work!