openrazer/openrazer #1466: Huntsman Mini support is half broken

Broeckelmann, 2021-03-22:
### Distribution
Fedora 33

### Support

- [X ] Have you tried rebooting?
- [X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I just installed the new razer packages and new driver. Now I have Huntsman Mini with lightning support, the lightning now works fine, but the fn-button no longer works, means no F5, etc works anymore. It worked before so it has to be new driver or something in the packages, I don't know which though. Would be nice if it can be fixed, because without the fn support it's kind of a useless keyboard ^-^.

z3ntu, 2021-03-22:
As a workaround you can stop the daemon (`systemctl --user stop openrazer-daemon.service`)

---

So essentially it seems at least these devices have broken Fn keys currently:
* Huntsman Mini - #1466
* Huntsman TE - https://github.com/openrazer/openrazer/pull/1363#issuecomment-786440971
* BlackWidow (2019) - #1459

Please try applying this patch to the driver source, compile and load this driver (`make driver && sudo rmmod razerkbd && sudo insmod ./driver/razerkbd.ko` should work) and give me the output from dmesg from it when using the Fn key?
```diff
diff --git a/driver/razerkbd_driver.c b/driver/razerkbd_driver.c
index fdf4569..c2cd895 100644
--- a/driver/razerkbd_driver.c
+++ b/driver/razerkbd_driver.c
@@ -2185,6 +2185,8 @@ static int razer_raw_event(struct hid_device *hdev, struct hid_report *report, u
         return 0;
     }
 
+    printk(KERN_ERR "razerkbd event: size %d - data\n", size);
+    print_hex_dump(KERN_ALERT, " razerkbd: ", DUMP_PREFIX_NONE, 16, 1, data, size, false);
     // The event were looking for is 16 or 22 bytes long and starts with 0x04.
     // Newer firmware seems to use 22 bytes.
     if(intf->cur_altsetting->desc.bInterfaceProtocol == USB_INTERFACE_PROTOCOL_KEYBOARD &&

```
**Also please only copy out the parts from your dmesg which are not from normal typing (and entering passwords..) because it's essentially a key logger :D**
(adapted from https://github.com/openrazer/openrazer/pull/873#issuecomment-577804267)

Broeckelmann, 2021-03-22:
Thank you very much for the quick answer, the workaround works (`systemctl --user stop openrazer-daemon.service`).

I'll try applying the patch next.

Broeckelmann, 2021-03-22:
So I tried around a bit:
`Huntsman Mini dmesg:
// This is a normal event
[  312.333553] razerkbd event: size 8 - data
[  312.333557]  razerkbd: 00 00 00 00 00 00 00 00

// These are fn events
[  313.030879] razerkbd event: size 22 - data
[  313.030890]  razerkbd: 02 e2 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  313.030892]  razerkbd: 00 00 00 00 00 00
[  313.121555] razerkbd event: size 22 - data
[  313.121560]  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  313.121562]  razerkbd: 00 00 00 00 00 00
[  313.314549] razerkbd event: size 22 - data
[  313.314553]  razerkbd: 02 ea 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  313.314556]  razerkbd: 00 00 00 00 00 00
[  313.432547] razerkbd event: size 22 - data
[  313.432550]  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  313.432552]  razerkbd: 00 00 00 00 00 00
[  313.600552] razerkbd event: size 22 - data
[  313.600556]  razerkbd: 02 e9 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  313.600558]  razerkbd: 00 00 00 00 00 00
[  313.714686] razerkbd event: size 22 - data
[  313.714691]  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  313.714693]  razerkbd: 00 00 00 00 00 00
[  313.869676] razerkbd event: size 22 - data
[  313.869680]  razerkbd: 02 b6 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  313.869682]  razerkbd: 00 00 00 00 00 00
[  313.984687] razerkbd event: size 22 - data
[  313.984690]  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  313.984692]  razerkbd: 00 00 00 00 00 00
[  314.174669] razerkbd event: size 22 - data
[  314.174673]  razerkbd: 02 cd 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  314.174675]  razerkbd: 00 00 00 00 00 00
[  314.312669] razerkbd event: size 22 - data
[  314.312672]  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  314.312674]  razerkbd: 00 00 00 00 00 00
[  314.499668] razerkbd event: size 22 - data
[  314.499671]  razerkbd: 02 b5 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  314.499672]  razerkbd: 00 00 00 00 00 00
[  314.604668] razerkbd event: size 22 - data
[  314.604671]  razerkbd: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  314.604673]  razerkbd: 00 00 00 00 00 00
[  315.342669] razerkbd event: size 22 - data
[  315.342672]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00
[  315.342674]  razerkbd: 00 00 00 00 00 00
[  315.487668] razerkbd event: size 22 - data
[  315.487671]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  315.487672]  razerkbd: 00 00 00 00 00 00
[  316.244670] razerkbd event: size 22 - data
[  316.244674]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00
[  316.244676]  razerkbd: 00 00 00 00 00 00
[  316.356669] razerkbd event: size 22 - data
[  316.356674]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  316.356675]  razerkbd: 00 00 00 00 00 00
[  316.516668] razerkbd event: size 22 - data
[  316.516669]  razerkbd: 01 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00
[  316.516670]  razerkbd: 00 00 00 00 00 00
[  316.629668] razerkbd event: size 22 - data
[  316.629670]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  316.629672]  razerkbd: 00 00 00 00 00 00
[  316.806667] razerkbd event: size 22 - data
[  316.806670]  razerkbd: 01 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00
[  316.806671]  razerkbd: 00 00 00 00 00 00
[  316.934669] razerkbd event: size 22 - data
[  316.934673]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  316.934675]  razerkbd: 00 00 00 00 00 00
[  317.084685] razerkbd event: size 22 - data
[  317.084687]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 08 00 00 00 00
[  317.084689]  razerkbd: 00 00 00 00 00 00
[  317.210669] razerkbd event: size 22 - data
[  317.210674]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  317.210675]  razerkbd: 00 00 00 00 00 00
[  317.350670] razerkbd event: size 22 - data
[  317.350674]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 04 00 00 00 00
[  317.350676]  razerkbd: 00 00 00 00 00 00
[  317.470669] razerkbd event: size 22 - data
[  317.470672]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  317.470673]  razerkbd: 00 00 00 00 00 00
[  317.793669] razerkbd event: size 22 - data
[  317.793672]  razerkbd: 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  317.793674]  razerkbd: 00 00 00 00 00 00
[  319.469557] razerkbd event: size 22 - data
[  319.469563]  razerkbd: 04 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  319.469565]  razerkbd: 00 00 00 00 00 00
[  319.798554] razerkbd event: size 22 - data
[  319.798558]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00
[  319.798559]  razerkbd: 00 00 00 00 00 00
[  319.920554] razerkbd event: size 22 - data
[  319.920558]  razerkbd: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  319.920560]  razerkbd: 00 00 00 00 00 00
[  320.588557] razerkbd event: size 22 - data
[  320.588563]  razerkbd: 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  320.588565]  razerkbd: 00 00 00 00 00 00
[  320.815553] razerkbd event: size 22 - data
[  320.815557]  razerkbd: 01 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00
[  320.815558]  razerkbd: 00 00 00 00 00 00
`

Broeckelmann, 2021-03-22:
Tell me when you need more for testing.

z3ntu, 2021-03-22:
Weird, with v3.0.0 this should work now on your device.. And even with this and the daemon started, the fn key doesn't work?

Broeckelmann, 2021-03-22:
Nop, I can try to reinstall and then see.

Broeckelmann, 2021-03-22:
Still doesn't works, hmm.

sebw, 2021-03-22:
Just upgraded and Fn key do not work on my Huntsman mini either.

z3ntu, 2021-03-23:
@fjpanag Can you please try the v3.0.0 release on your BlackWidow (2019) and check again if it works or doesn't work for you?

sebw, 2021-03-23:
#1468 works for me on Huntsman mini!

fjpanag, 2021-03-23:
@z3ntu I just tried v3.0.0, but still nothing...

z3ntu, 2021-03-24:
@sebw Thanks for confirming!

@fjpanag Can you please also patch the driver as written in https://github.com/openrazer/openrazer/issues/1466#issuecomment-804066364 and send the dmesg parts?

unphased, 2021-03-26:
I'm on a huntsman mini and stopping the service didn't fix my fn key...

ne04, 2021-03-27:
Also with Razer Huntsman mini, fn key isn't working. This means no access to F1, F2, etc, as well as arrow keys and volume control, among other things.

unphased, 2021-03-27:
I'm definitely returning my huntsman mini if this doesnt work soon. because the default position of the fn key and arrow keys is unacceptable, however the key switches are SO GOOD. Very torn I am. Will Openrazer have the capability of changing which key is Fn key? Will openrazer be able to write to the keyboard to make that capability possible to use under windows? I'd like to make it so that right Alt is fn key, and just use shift, ctrl, menu, and fn keys as arrow keys. That's the first step. Then the next step is to enable sane access to page up/down.

closed 2021-03-27 by z3ntu

fjpanag, 2021-04-01:
I'm really sorry for the late reply.

It seems that on latest on master (47b79ff1c86666496ead7c36e3791c39a9c0407c), everything works correctly!  
Thank you!

kpuljek, 2021-04-10:
I just got the update to 3.0.1 and the Fn key still doesn't work. Do I have to do anything else?

z3ntu, 2021-04-10:
Have you rebooted after updating? Also to confirm, you're using the Huntsman Mini, right?

kpuljek, 2021-04-10:
Yeah, Huntsman Mini, and have just removed everything and reinstalled it again, then rebooted.
openrazer-daemon --version
openrazer-daemon 3.0.1

Still nothing. :/

alimagik1, 2022-10-30:
Distribution
Arch Linux

Support
[X ] Have you tried rebooting?
[X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Kernel ver 6.0.5 and 5.15.75-1

BlackWidow V3 Mini HyperSpeed also found a bug in the Fn key in the Japanese version

In the Japanese version, pressing the Fn key to switch between English and Japanese works as an ECS key, but if openrazer-daemon.service is running, it does not work as an ECS key. Works as a key to switch between Japanese and English. (For the English version, the ECS key and the ~ key)

Only one key has a problem. Keys that combine with other Fn keys work.

If you do one of the following as a temporary measure, it will work normally.
1.Stop openrazer-daemon.service
2.Uninstall kernel-header
3.Connect the keyboard with Bluetooth

hope it improves

z3ntu, 2022-10-30:
@alimagik1 I don't quite understand what you mean.

From what I can see from pictures of the keyboard (and openrazer code), at least the US version has ` and ~ characters on the Esc key. So normally you get Esc key, and when you press Fn+Esc then you get one of these characters.

How it this different from your variant, the Japanese version? Might be useful to include a picture of your keyboard so I understand the problem better. I also have no clue about input with Japanese so "key to switch between Japanese and English" doesn't say anything to me.

alimagik1, 2022-10-30:
> @alimagik1 I don't quite understand what you mean.
>
> From what I can see from pictures of the keyboard (and openrazer code), at least the US version has ` and ~ characters on the Esc key. So normally [...]

thanks for reply

Paste the image of the Japanese keyboard layout

[image]

In the Japanese version, a key to switch between English and Japanese is assigned to the position of the ECS key in the English version.

If I press the English/Japanese switch key at the same time as the Fn key, I want it to work as an ESC key, but the English/Japanese switch key doesn't change.

When you stop the operation of OpenRazer, pressing the Fn key and the Japanese/English switching key will work as the ESC key.

Other keys work fine when pressed simultaneously with the Fn key.
Pressing the Fn key + 1 key works as F1.

Thank you

z3ntu, 2022-10-30:
Can you run `sudo libinput debug-events --show-keycodes` (you might need to install some extra packages so this command works) and press this button (when it does work for switching) and send the output here?

alimagik1, 2022-10-31:
When running openrazer-daemon.service results in not recognizing the Fn key

```
sudo libinput debug-events --show-keycodes
-event2   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event8   DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event1   DEVICE_ADDED            Lid Switch                        seat0 default group3  cap:S
-event0   DEVICE_ADDED            Sleep Button                      seat0 default group4  cap:k
-event6   DEVICE_ADDED            Integrated Camera: Integrated C   seat0 default group5  cap:k
-event7   DEVICE_ADDED            Integrated Camera: Integrated I   seat0 default group5  cap:k
-event27  DEVICE_ADDED            Razer Razer BlackWidow V3 Mini    seat0 default group6  cap:k
-event28  DEVICE_ADDED            Razer Razer BlackWidow V3 Mini    seat0 default group6  cap:kp scroll-nat
-event29  DEVICE_ADDED            Razer Razer BlackWidow V3 Mini    seat0 default group6  cap:p left scroll-nat scroll-button
-event30  DEVICE_ADDED            Razer Razer BlackWidow V3 Mini    seat0 default group6  cap:k
-event25  DEVICE_ADDED            ELAN0670:00 04F3:3150 Mouse       seat0 default group7  cap:p left scroll-nat scroll-button
-event26  DEVICE_ADDED            ELAN0670:00 04F3:3150 Touchpad    seat0 default group7  cap:pg  size 99x54mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on
-event12  DEVICE_ADDED            HDA Intel PCH Headphone           seat0 default group8  cap:k
-event4   DEVICE_ADDED            Intel HID events                  seat0 default group9  cap:k
-event3   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group10 cap:k
-event10  DEVICE_ADDED            TPPS/2 JYT_Synaptics TrackPoint   seat0 default group11 cap:p left scroll-nat scroll-button
-event5   DEVICE_ADDED            ThinkPad Extra Buttons            seat0 default group12 cap:k
-event31  DEVICE_ADDED            DA V2 Pro Mouse                   seat0 default group13 cap:p left scroll-nat scroll-button
-event256  DEVICE_ADDED            DA V2 Pro Consumer Control        seat0 default group13 cap:kp scroll-nat
-event257  DEVICE_ADDED            DA V2 Pro System Control          seat0 default group13 cap:k
-event259  DEVICE_ADDED            DA V2 Pro Keyboard                seat0 default group13 cap:k
-event28  KEYBOARD_KEY            +0.000s	KEY_GRAVE (41) pressed
 event28  KEYBOARD_KEY            +0.093s	KEY_GRAVE (41) released
 event28  KEYBOARD_KEY            +3.659s	KEY_GRAVE (41) pressed
 event28  KEYBOARD_KEY            +3.785s	KEY_GRAVE (41) released

```

Fn key works if openrazer-daemon.service is stopped

```
sudo libinput debug-events --show-keycodes
-event2   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event8   DEVICE_ADDED            Video Bus                         seat0 default group2  cap:k
-event1   DEVICE_ADDED            Lid Switch                        seat0 default group3  cap:S
-event0   DEVICE_ADDED            Sleep Button                      seat0 default group4  cap:k
-event6   DEVICE_ADDED            Integrated Camera: Integrated C   seat0 default group5  cap:k
-event7   DEVICE_ADDED            Integrated Camera: Integrated I   seat0 default group5  cap:k
-event27  DEVICE_ADDED            Razer Razer BlackWidow V3 Mini    seat0 default group6  cap:k
-event28  DEVICE_ADDED            Razer Razer BlackWidow V3 Mini    seat0 default group6  cap:kp scroll-nat
-event29  DEVICE_ADDED            Razer Razer BlackWidow V3 Mini    seat0 default group6  cap:p left scroll-nat scroll-button
-event30  DEVICE_ADDED            Razer Razer BlackWidow V3 Mini    seat0 default group6  cap:k
-event25  DEVICE_ADDED            ELAN0670:00 04F3:3150 Mouse       seat0 default group7  cap:p left scroll-nat scroll-button
-event26  DEVICE_ADDED            ELAN0670:00 04F3:3150 Touchpad    seat0 default group7  cap:pg  size 99x54mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on
-event12  DEVICE_ADDED            HDA Intel PCH Headphone           seat0 default group8  cap:k
-event4   DEVICE_ADDED            Intel HID events                  seat0 default group9  cap:k
-event3   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group10 cap:k
-event10  DEVICE_ADDED            TPPS/2 JYT_Synaptics TrackPoint   seat0 default group11 cap:p left scroll-nat scroll-button
-event5   DEVICE_ADDED            ThinkPad Extra Buttons            seat0 default group12 cap:k
-event31  DEVICE_ADDED            DA V2 Pro Mouse                   seat0 default group13 cap:p left scroll-nat scroll-button
-event256  DEVICE_ADDED            DA V2 Pro Consumer Control        seat0 default group13 cap:kp scroll-nat
-event257  DEVICE_ADDED            DA V2 Pro System Control          seat0 default group13 cap:k
-event259  DEVICE_ADDED            DA V2 Pro Keyboard                seat0 default group13 cap:k
-event27  KEYBOARD_KEY            +0.000s	KEY_ESC (1) pressed
^[ event27  KEYBOARD_KEY            +0.125s	KEY_ESC (1) released
 event27  KEYBOARD_KEY            +2.637s	KEY_GRAVE (41) pressed
 event27  KEYBOARD_KEY            +2.755s	KEY_GRAVE (41) released
 event27  KEYBOARD_KEY            +3.788s	KEY_ESC (1) pressed
^[ event27  KEYBOARD_KEY            +3.929s	KEY_ESC (1) released
 event27  KEYBOARD_KEY            +5.162s	KEY_GRAVE (41) pressed
[... cut ...]
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```

Is there anything else I can do?

alimagik1, 2022-11-06:
When I updated the system with the yay command on November 6, 2022 Japan time, I confirmed that the Japanese version of BlackWidow V3 Mini HyperSpeed can switch between the Japanese-English switch key and the ESC key correctly.

kernel version that worked
linux kernel ver 6.07 and 5.15.77

Thank you very much for your help.

bathwaterpizza, 2024-01-24:
Running openrazer v3.7 with polychromatic on Manjaro, Fn keys are not working and there is no option to remap the keys in multiple frontends I've tried.

PR #1468 (2021-03-27): Add fn translate keys for huntsman mini keyboard
Translate keys are different in the hunstman mini since it is a 60% keyboard, the other translations keys do not work with this keyboard.

Fixes #1466