openrazer/openrazer #1467: Support for Webcam Razer Kiyo

eddiecsilva, 2021-03-22:
### Device Info

* Name: Razer Kiyo
* Product Number: 
* Model Number: RZ19-0232

### Device Information
```
Please post the output of

Bus 001 Device 002: ID 1532:0e03 Razer USA, Ltd Gaming Webcam [Kiyo]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0e03 Gaming Webcam [Kiyo]
  bcdDevice           79.13
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x02c7
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              0 
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength       0x0069
        dwClockFrequency       30.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             4
        wTerminalType      0x0101 USB Streaming
[... 475 lines cut ...]
```

```
Output of ls -lah /dev/input/by-id/

lrwxrwxrwx 1 root root   9 mar 22 09:51 usb-Alpha_Imaging_Tech._Corp._Razer_Kiyo-event-if00 -> ../event9
lrwxrwxrwx 1 root root   9 mar 22 09:51 usb-GeneralPlus_USB_Audio_Device-event-if03 -> ../event7
lrwxrwxrwx 1 root root   9 mar 22 09:51 usb-Microsoft_SideWinder™_Mouse-event-joystick -> ../event3
lrwxrwxrwx 1 root root   9 mar 22 09:51 usb-Microsoft_SideWinder™_Mouse-event-mouse -> ../event2
lrwxrwxrwx 1 root root   6 mar 22 09:51 usb-Microsoft_SideWinder™_Mouse-joystick -> ../js0
lrwxrwxrwx 1 root root   9 mar 22 09:51 usb-Microsoft_SideWinder™_Mouse-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 mar 22 09:51 usb-Microsoft_Wired_Keyboard_600-event-if01 -> ../event5
lrwxrwxrwx 1 root root   9 mar 22 09:51 usb-Microsoft_Wired_Keyboard_600-event-kbd -> ../event4
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
[image]
[image]
[image]
[image]

z3ntu, 2021-03-22:
Please see #1328

closed 2021-03-22 by z3ntu

eddiecsilva, 2021-03-22:
I apologize for the duplicate issue, I only researched the open ones and forgot to look at the closed ones. 
Thanks for the information.