openrazer/openrazer #1472: OpenRazer randomly crashes

Mythbusters123, 2021-03-25:
### Distribution
Arch Linux

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? yes

Ever since 3.0.0 came out, I am not able to change any keys on my blackwidow elite. I have looked at the logs and all it says is that its resuming the keyboard. This happens randomly for me, and sometimes when I open discord it just randomly stops. I have tried using polychromatic and razerCommander. I have also trioed switching kernles from Zen to Lts, to no avail. Later today I will try razerGenie/Snake, and possibly reinstall the drivers again. Yes, my user is in the plugdev group and yes my keyboard is connected :joy:

z3ntu, 2021-03-26:
> I am not able to change any keys on my blackwidow elite

You mean you aren't able to change the lighting colors of your keys?

Mythbusters123, 2021-03-26:
Yea, sorry for poor wording

z3ntu, 2021-03-26:
The device is shown in the the various frontends, yes?
Can you try downgrading to v2.9.0 to see if that changes anything?

Mythbusters123, 2021-03-26:
How would I go about downgrading? I am running arch Linux if that is helpful.

z3ntu, 2021-03-26:
If you still have the old .pkg.tar.zst lying around, you can just do `sudo pacman -U $thatfile` . Alternatively you can check out the commit in the AUR repo that points to v2.9.0 and run makepkg there.

Mythbusters123, 2021-03-29:
I just rolled back to 2.9, and nothing fixed it.

Mythbusters123, 2021-03-31:
Reinstalled arch completely. Still didnt help.

Mythbusters123, 2021-04-08:
Alright, so digging into they tty said that the report length was invalid and gave the number -100

z3ntu, 2021-04-08:
Not sure what's happening there...

z3ntu, 2021-04-14:
Sounds like an issue either with your computer or your keyboard. Can you test the keyboard on a different PC whether it works correctly there?

Mythbusters123, 2021-04-19:
I dual boot windows and linux, and razer synapse works perfectly fine on windows, also I think 3.0.1 fixed this but let me do more testing on this

closed 2021-09-17 by github-actions[bot]