openrazer/openrazer #1480: Support for BlackWidow Tournament Edition Chroma V2

vinolivae, 2021-03-30:
Name: BlackWidow Tournament Edition Chroma V2
Product N: RZ03-02190700-R3M1

### Device Info
```
~ % lsusb -d 1532: -v

Bus 001 Device 010: ID 1532:0209 Razer USA, Ltd BlackWidow Tournament Edition Chroma
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0209 BlackWidow Tournament Edition Chroma
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow Tournament Edition Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```
~ % ls -la /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 260 mar 30 11:06 .
drwxr-xr-x 4 root root 820 mar 30 11:06 ..
lrwxrwxrwx 1 root root  10 mar 30 11:06 usb-18f8_USB_GAMING_MOUSE-event-if01 -> ../event12
lrwxrwxrwx 1 root root   9 mar 30 11:06 usb-18f8_USB_GAMING_MOUSE-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 mar 30 11:06 usb-18f8_USB_GAMING_MOUSE-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root   9 mar 30 11:06 usb-18f8_USB_GAMING_MOUSE-mouse -> ../mouse3
lrwxrwxrwx 1 root root   9 mar 30 10:54 usb-CKFIH56Q34442002B3D0_Integrated_Webcam_HD_200901010001-event-if00 -> ../event4
lrwxrwxrwx 1 root root   9 mar 30 10:54 usb-CKFIH56Q34442002B3D0_Integrated_Webcam_HD_200901010001-event-if02 -> ../event6
lrwxrwxrwx 1 root root  10 mar 30 11:06 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-event-if01 -> ../event17
lrwxrwxrwx 1 root root  10 mar 30 11:06 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 mar 30 11:06 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if01-event-kbd -> ../event14
lrwxrwxrwx 1 root root  10 mar 30 11:06 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if02-event-mouse -> ../event18
lrwxrwxrwx 1 root root   9 mar 30 11:06 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if02-mouse -> ../mouse4

```

z3ntu, 2021-05-08:
The device is already supported for a long time. Does it not work for you?

isfalcantara, 2021-05-25:
This specific keyboard isn't listed under the supported devices. Yes, the BlackWidow Chroma V2 or the BlackWidow Chroma Tournament edition are, but not the BlackWidow Chroma **V2** Tournament Edition. Is this intentional?

vinolivae, 2021-05-25:
Hi @z3ntu, thanks for replying. Apologies for the delay. :)

It did not work. Whenever I opened RazerGenie he said that my device was not supported and that I should check the list.

Today I installed Polychromatic and it worked perfectly.

closed 2021-05-29 by vinolivae