openrazer/openrazer #1481: Python error in "/usr/bin/openrazer-daemon", line 140

noche-x, 2021-03-31:
### Distribution
Arch Linux 5.11.10-arch1-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I tried to run `openrazer-daemon -Fv` and it gave me `TypeError: __init__() got an unexpected keyword argument 'persistence_file'`. I looked around in GitHub and saw that the parameters match and there shouldn't be any errors technically. I switched to *-git versions from the AUR but no luck.

I first encountered this issue a couple of days ago when I accidentally changed my dpi on my mouse and when I tried to look at polychromatic-controler but it said it didn't recognize my devices. I used a previous version but no luck, it said openrazer-daemon not running or something like that. I don't know if this is caused by a python package update unfortunately.

lah7, 2021-03-31:
Here's a few things you can try:

1. Try deleting the configuration in `~/.config/openrazer/` and see if the daemon runs again.  Frontends won't run without `openrazer-daemon` running.

2. It sounds like a mix and match of versions are installed. Make sure you haven't got another copy of OpenRazer accidentally installed in `/usr/local/bin`, `/usr/local/lib`, etc.

3. This command will list Python packages installed for an older Python version - if so, you'll need to reinstall them:

       pacman -Qo /usr/lib/python3.8/

noche-x, 2021-03-31:
1. Deleted the whole folder, no difference in the error.
2. Wasn't able to find any sort of `openrazer` in `/usr/local/bin` and `/usr/local/lib`
3. `openrazer-daemon-git` and `python-openrazer-git` owns `/usr/lib/python3.9`

noche-x, 2021-03-31:
I tried to cleanBuild but made no difference. Going to restart in a couple of minutes to see if that helps.

noche-x, 2021-03-31:
I fixed the issue thanks to z3ntu over at the matrix chat. 

We diagnosed the problem was that I had a mismatch of /usr/lib/python3.9/site-packages/openrazer_daemon/daemon.py and /usr/bin/openrazer-daemon.

With `find / -name openrazer_daemon` I saw that I had multiple versions:
```
/usr/lib/python3.9/openrazer_daemon
/usr/lib/python3.9/site-packages/openrazer_daemon
```

I ran `pacman -Qo` for both paths and the first one wasn't owned by anything. So I just removed it and it was fine.