openrazer/openrazer #1482: Debian sid - invalid/expired pubkey 2021-03-28 (fix inside)

dpanter, 2021-04-04:
For those running an older install, you may have split key storage, or added keys with **apt-key** _(deprecated in favor of **gpg**)_ .
You may see errors like this during apt update:
`W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://download.opensuse.org/repositories/hardware:/razer/Debian_Unstable  InRelease: The following signatures were invalid: EXPKEYSIG 062F9FD4D6D11CE4 hardware OBS Project <hardware@build.opensuse.org>`

Trying to add the new gpg key as instructed on OBS does not solve the issue. The older key has priority and must be deleted first. Check keys with **apt-key** and delete the old **expired** key. Then add new **gpg** key and update.
Example on my system, find and remove old key:

```
$ apt-key list
...
/etc/apt/trusted.gpg
pub   rsa2048 2014-09-16 [SC] [expired: 2021-03-28]
      E96C 496E 4E77 C159 C2FC  A053 062F 9FD4 D6D1 1CE4
uid           [ expired] hardware OBS Project <hardware@build.opensuse.org>
```
```
$ sudo apt-key del E96C496E4E77C159C2FCA053062F9FD4D6D11CE4
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
OK
```

Add new key and check that it works:
```
$ curl -fsSL https://download.opensuse.org/repositories/hardware:razer/Debian_Unstable/Release.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/hardware_razer.gpg > /dev/null
```
```
$ sudo apt update
...
Get:16 http://download.opensuse.org/repositories/hardware:/razer/Debian_Unstable  InRelease [1 555 B]
```
Check new/correct key location:
```
$ apt-key list
...
/etc/apt/trusted.gpg.d/hardware_razer.gpg
-----------------------------------------
pub   rsa2048 2014-09-16 [SC] [expires: 2023-05-23]
      E96C 496E 4E77 C159 C2FC  A053 062F 9FD4 D6D1 1CE4
uid           [ unknown] hardware OBS Project <hardware@build.opensuse.org>

```

Perhaps this is a good time to double check your other keys as well and do some spring cleaning, and stop using apt-key... ;)
For more information, check out the manpages, Debian wiki or this [LinuxUprising article](https://www.linuxuprising.com/2021/01/apt-key-is-deprecated-how-to-add.html)

z3ntu, 2021-04-04:
I've pinned this, thanks for the write-up!

closed 2021-04-04 by z3ntu

velespr0, 2021-04-10:
Thanks for the help

muhuk, 2021-05-17:
Just updated razer repository from deb9 to deb10. I seem to have the correct key, but I still get the following error:

```
Err:6 http://download.opensuse.org/repositories/hardware:/razer/Debian_10  InRelease
  The following signatures were invalid: EXPKEYSIG 062F9FD4D6D11CE4 hardware OBS Project <hardware@build.opensuse.org>
Hit:13 https://updates.signal.org/desktop/apt xenial InRelease
Reading package lists... Done
W: GPG error: http://download.opensuse.org/repositories/hardware:/razer/Debian_10  InRelease: The following signatures were invalid: EXPKEYSIG 062F9FD4D6D11CE4 hardware OBS Project <hardware@build.opensuse.org>
E: The repository 'http://download.opensuse.org/repositories/hardware:/razer/Debian_10  InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
```