openrazer/openrazer #1483: Support for Razer Kraken X headphones

Kostas-Mavrogalos, 2021-04-05:
### Distribution

Distributor:	Ubuntu, and more specifically, xUbuntu
Release:	20.04

Hey, I came here through a small quest with Razer. Is it possible to make Razer Kraken X headphones have support with this awesome app? I would love to have it and it's not yet supported

z3ntu, 2021-04-05:
I think these headphones are covered by this comment https://github.com/openrazer/openrazer/issues/1396#issuecomment-753455216

Kostas-Mavrogalos, 2021-04-06:
It would, but mine don't have a jack, but a USB. Also, OpenRazer says it knows that a Razer product is connected to the PC, but the daemon doesn't detect it, so idk :/

z3ntu, 2021-04-06:
Ok, can you send the output of `lsusb | grep 1532` please?

Kostas-Mavrogalos, 2021-04-06:
Bus 001 Device 006: ID 1532:0526 Razer USA, Ltd Razer Kraken X USB

z3ntu, 2021-04-06:
The device is tracked in #1326

closed 2021-04-06 by z3ntu