openrazer/openrazer #1484: BlackWidow Chroma forgets custom layout after reboot/replug since 3.0 update

Liemaeu, 2021-04-06:
### Distribution

Kubuntu 20.04

### Support

- [y] Have you tried rebooting?
- [y] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Since the update to version 3.0 when I reboot my pc or plug my BlackWidow Chroma keyboard out and in, it forgets the custom layout and turns completely in green. This happens with all programs (e.g. RazerGenie, razerCommander), so I think this is an openrazer releated issue.
This never happend on older openrazer versions.

lah7, 2021-04-07:
OpenRazer 3.0 introduced a "persistence" feature, which does indeed restore the last hardware effect to assist newer Razer hardware that were incapable of retaining their effects across reboots.

At the moment, this feature doesn't _persist_ custom effects nor can the feature be disabled entirely. It's up to a script (or frontend) to trigger the custom layout at login.

A workaround for now could be to patch the daemon so no hardware effect is applied at start up:

    sudo sed -i 's/self.restore_effect()/#self.restore_effect()/g' /usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py

_(The path will differ for non-Ubuntu/Debian users)_

lah7, 2021-04-07:
I propose #1485 to add a config key so restoring effects at startup can be disabled off in `~/.config/openrazer/razer.conf` to restore previous <=2.9.0 behaviour.

Liemaeu, 2021-04-07:
> OpenRazer 3.0 introduced a "persistence" feature, which does indeed restore the last hardware effect to assist newer Razer hardware that were incapable of retaining their effects across reboots.
>
> At [...]

Worked! Thank you very much.

PR #1485 (2021-04-10): Add config key "restore_persistence"
No changes to the default behaviour. This allows users to optionally turn off persistence at startup.

Intended for use cases where devices should not be touched at start up. For instance, effects are handled by another program, a script or to retain the last frame from a matrix effect across reboots (useful for older firmware like BlackWidow Chroma)

Fixes #1484.
closed 2021-04-10 by z3ntu