openrazer/openrazer #1490: Support for Kraken X USB [1532:0526]

z3ntu, 2021-04-10:
*Re-opened from #1326 because GitHub search is incompetent and can't find issues that were opened by users that have been deleted in the meantime.*

### Device Info

* Name: Razer USA, Ltd Razer Kraken X USB
* Product Number: RZ04-02960100-R3U1
* Model Number: RZ04-0296

### Device Information
```
Bus 001 Device 003: ID 1532:0526 Razer USA, Ltd Razer Kraken X USB
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0526 
  bcdDevice            0.34
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken X USB
  iSerial                 3 00000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00fa
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x004f
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
[... 196 lines cut ...]
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     109
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0025  1x 37 bytes
        bInterval               1
```

```
 usb-Razer_Razer_Kraken_X_USB_00000000-event-if03
```

IzNoGud78, 2021-04-17:
Hope it can be added soon to the list of supported devices.

M2rsho, 2021-12-16:
Still waiting

lucifer-crybaby, 2022-01-11:
any update on this one ?

RiemaruKarurosu, 2022-07-10:
where could I upload the pcaps? or there's no need of them?

z3ntu, 2022-07-23:
@RiemaruKarurosu As attachment to a comment here, or Google Drive or something. And yes they'd be useful so I can give some pointers how the implementation should be like.