openrazer/openrazer #1492: Couldn't convert event to key name

khasanovbi, 2021-04-11:
RazerBlade2019Adv
Ubuntu 20.04.2 LTS

See some errors like this in `~/.local/share/openrazer/logs/razer.log`:

```python
2021-04-04 15:18:49 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 126
```

```
➜  ~ cat ~/.local/share/openrazer/logs/razer.log | grep -i KeyError | sort | uniq
KeyError: 126
KeyError: 224
KeyError: 225
```

z3ntu, 2021-05-08:
Do you know what you have done when this happened? Seems to be related to some macro code, have you used that?

ttsesm, 2021-06-06:
I get a similar error when I try to play with the brightness of the screen and using the macro combination of `fn+F8/9`:

```
Jun 06 15:38:24 blade openrazer-daemon[2101]: 2021-06-06 15:38:24 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Jun 06 15:38:24 blade openrazer-daemon[2101]: Traceback (most recent call last):
Jun 06 15:38:24 blade openrazer-daemon[2101]:   File "/usr/lib/python3.9/site-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
Jun 06 15:38:24 blade openrazer-daemon[2101]:     key_name = self.EVENT_MAP[key_id]
Jun 06 15:38:24 blade openrazer-daemon[2101]: KeyError: 224
Jun 06 15:38:24 blade root[4906]: ACPI group/action undefined: video/brightnessdown / BRTDN
Jun 06 15:38:24 blade dbus-daemon[1250]: [system] Activating service name='org.kde.powerdevil.backlighthelper' requested by ':1.43' (uid=1000 pid=2281 comm="/usr/lib/org_kde_powerdevil ") (using servicehelper)
Jun 06 15:38:24 blade dbus-daemon[1250]: [system] Successfully activated service 'org.kde.powerdevil.backlighthelper'
Jun 06 15:38:24 blade openrazer-daemon[2101]: 2021-06-06 15:38:24 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Jun 06 15:38:24 blade openrazer-daemon[2101]: Traceback (most recent call last):
Jun 06 15:38:24 blade openrazer-daemon[2101]:   File "/usr/lib/python3.9/site-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
Jun 06 15:38:24 blade openrazer-daemon[2101]:     key_name = self.EVENT_MAP[key_id]
Jun 06 15:38:24 blade openrazer-daemon[2101]: KeyError: 224
Jun 06 15:38:24 blade openrazer-daemon[2101]: 2021-06-06 15:38:24 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Jun 06 15:38:24 blade openrazer-daemon[2101]: Traceback (most recent call last):
Jun 06 15:38:24 blade openrazer-daemon[2101]:   File "/usr/lib/python3.9/site-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
Jun 06 15:38:24 blade openrazer-daemon[2101]:     key_name = self.EVENT_MAP[key_id]
Jun 06 15:38:24 blade openrazer-daemon[2101]: KeyError: 225
Jun 06 15:38:24 blade root[4956]: ACPI group/action undefined: video/brightnessup / BRTUP
Jun 06 15:38:24 blade openrazer-daemon[2101]: 2021-06-06 15:38:24 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Jun 06 15:38:24 blade openrazer-daemon[2101]: Traceback (most recent call last):
Jun 06 15:38:24 blade openrazer-daemon[2101]:   File "/usr/lib/python3.9/site-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
Jun 06 15:38:24 blade openrazer-daemon[2101]:     key_name = self.EVENT_MAP[key_id]
Jun 06 15:38:24 blade openrazer-daemon[2101]: KeyError: 225
Jun 06 15:38:24 blade root[4990]: ACPI group/action undefined: video/brightnessup / BRTUP
Jun 06 15:38:24 blade openrazer-daemon[2101]: 2021-06-06 15:38:24 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Jun 06 15:38:24 blade openrazer-daemon[2101]: Traceback (most recent call last):
Jun 06 15:38:24 blade openrazer-daemon[2101]:   File "/usr/lib/python3.9/site-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
Jun 06 15:38:24 blade openrazer-daemon[2101]:     key_name = self.EVENT_MAP[key_id]
Jun 06 15:38:24 blade openrazer-daemon[2101]: KeyError: 225
Jun 06 15:38:24 blade openrazer-daemon[2101]: 2021-06-06 15:38:24 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Jun 06 15:38:24 blade openrazer-daemon[2101]: Traceback (most recent call last):
Jun 06 15:38:24 blade openrazer-daemon[2101]:   File "/usr/lib/python3.9/site-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
Jun 06 15:38:24 blade openrazer-daemon[2101]:     key_name = self.EVENT_MAP[key_id]
Jun 06 15:38:24 blade openrazer-daemon[2101]: KeyError: 225
Jun 06 15:38:25 blade openrazer-daemon[2101]: 2021-06-06 15:38:25 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Jun 06 15:38:25 blade openrazer-daemon[2101]: Traceback (most recent call last):
Jun 06 15:38:25 blade openrazer-daemon[2101]:   File "/usr/lib/python3.9/site-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
Jun 06 15:38:25 blade openrazer-daemon[2101]:     key_name = self.EVENT_MAP[key_id]
Jun 06 15:38:25 blade openrazer-daemon[2101]: KeyError: 225
Jun 06 15:38:25 blade root[5024]: ACPI group/action undefined: video/brightnessup / BRTUP
Jun 06 15:38:25 blade openrazer-daemon[2101]: 2021-06-06 15:38:25 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Jun 06 15:38:25 blade openrazer-daemon[2101]: Traceback (most recent call last):
Jun 06 15:38:25 blade openrazer-daemon[2101]:   File "/usr/lib/python3.9/site-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
Jun 06 15:38:25 blade openrazer-daemon[2101]:     key_name = self.EVENT_MAP[key_id]
Jun 06 15:38:25 blade openrazer-daemon[2101]: KeyError: 225
```

For adjusting the brightness of the screen since I have an OLED screen I am using the [icc-brightness](https://github.com/udifuchs/icc-brightness) external application.

CreMindES, 2021-06-11:
I'm seeing the same errors while adjusting the brightness as well on a Razer Blade 15 Advanced 2020 non-OLED, fullHD  (EndeavourOS).

z3ntu, 2021-06-20:
Is there something not working or is this just a cosmetic error?

khasanovbi, 2021-06-20:
I have a problem with the keyboard in ubuntu, periodically it stops working for a short period of time. I don't know for sure if it's related to this or not.

z3ntu, 2021-06-20:
Could you maybe try uninstalling openrazer temporarily and see if this problem persists? Then we can rule out that openrazer causes this issue.

closed 2021-11-19 by github-actions[bot]