openrazer/openrazer #1495: Kernel update

Predrag, 2021-04-18:
Hello,

I have question. I am using Fedora 33 with current kernel version 5.11.14-200 with openrazer-driver version 3.0.1. Everything is working perfectly as expected. But I have small problem when kernel is upgraded from example 5.11.13-200 to 5.11.14-200 I have to manualy recompile openrazer-driver with this command sudo dkms install openrazer-driver/3.0.1 as in this [article](https://github.com/openrazer/openrazer/wiki/Troubleshooting). Would it be possible to somehow make it automated in the future ?

Thank you

z3ntu, 2021-05-08:
Might be fixed by https://github.com/openrazer/OBS-packaging/pull/4

z3ntu, 2021-05-08:
Please upgrade to the latest openrazer build (dnf upgrade) and check with the next kernel update if it broke again - I think it's fixed now. I'll close for now but we can re-open if the issue still exists.

closed 2021-05-08 by z3ntu