openrazer/openrazer #1497: OpenRazer DKMS SecureBoot Error

ScottMorris, 2021-04-20:
### Distribution

Linux Mint 20.1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm trying to setup OpenRazer on my install.  It used to be setup with my old install when I was running Linux Mint 19.3 and everything worked.  Since I've upgraded something weird seems to have happened and my Huntsman keyboard isn't working.  Looking into the issue more it appears that the OpenRazer DKMS driver thinks that SecureBoot is enabled, even when it isn't.

[image]

Upon following the linked troubleshooting guide here are the results of the listed command.

```
$ sudo mokutil --sb-state
This system doesn't support Secure Boot
```

Based on the results of this command I'm confused why SecureBoot is detected to be enabled when it isn't.

My device is the Razer Huntsman (1532:0227).

I'd be happy to provide additional information, as required.

lah7, 2021-04-20:
It could be a bug in Polychromatic's troubleshooter for OpenRazer, since the DKMS module says it's already loaded so secure boot may be irrelevant.

The [code at the moment](https://github.com/polychromatic/polychromatic/blob/dc95b9920b9cc4d7e8730978003fda3fc8b538c7/pylib/troubleshoot/openrazer.py#L124-L139) runs a command and [checks the last digit](https://wiki.archlinux.org/index.php/Unified_Extensible_Firmware_Interface/Secure_Boot#After_booting_the_OS):

    od --address-radix=n --format=u1 /sys/firmware/efi/efivars/SecureBoot*

On my system, this outputs:

       6   0   0   0   0

What does your system output?

Also, presumably, the Razer Huntsman isn't appearing at all within the application? Clear the log and let's take a look at the daemon's output:

```
rm ~/.local/share/openrazer/logs/razer.log
openrazer-daemon -s
openrazer-daemon -F
```

ScottMorris, 2021-04-21:
I've done some digging based on what you've mentioned and here is what I found.

I checked out the `ob` command but it returns an error, as `/sys/firmware/efi/efivars/SecureBoot*` related files aren't present on my machine.

The exact output is 

```
$ od --address-radix=n --format=u1 /sys/firmware/efi/efivars/SecureBoot*
od: '/sys/firmware/efi/efivars/SecureBoot*': Invalid argument
```

I looked into the code [you pointed to](https://github.com/polychromatic/polychromatic/blob/dc95b9920b9cc4d7e8730978003fda3fc8b538c7/pylib/troubleshoot/openrazer.py#L124-L139) and this should be fine, as there is a check for if the files exist before it tries to runs the `ob` command.

What is interesting though is the `passed` section of the result object only checks if the `status` is `0` (`"passed": True if int(status) == 0 else False`) and will return `True`, which I'm guessing means the check passed.  When the files aren't present, it [explicitly sets](https://github.com/polychromatic/polychromatic/blob/dc95b9920b9cc4d7e8730978003fda3fc8b538c7/pylib/troubleshoot/openrazer.py#L133) the status to `1` with a nice comment.

```py
# Likely secure boot is not present
status = 1
```

If I manually modify this file on my system and set `status = 0` then it is happy with this check.  This is still a _polychromatic_ thing and not an _openrazer_ thing so I kept looking.

```py
# Likely secure boot is not present
status = 0
```

I then went on to checkout the `openrazer-daemon` to see what it was up to.  It appears your hunch was right and my Huntsman isn't showing up in the logs. 

Here is what I got,

```
$ openrazer-daemon -F
Starting daemon.
2021-04-20 21:02:40 | razer                          | INFO     | Initialising Daemon (v2.5.0). Pid: 9648
2021-04-20 21:02:40 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-04-20 21:02:40 | razer                          | INFO     | Serving DBus
```

I decided to check `lsusb` to see if the device was showing up there, and it was,

```
$ lsusb | grep Razer
Bus 005 Device 012: ID 1532:0227 Razer USA, Ltd
```

I tried unplugging and plugging in my keyboard with no updates to the `openrazer-daemon` logs.

lah7, 2021-04-21:
I see the problem -- you're running 2.5.0 from the Ubuntu repositories. I don't think the Huntsman was supported during that time.

PPAs get disabled during upgrades so you'll need to re-enable them via Software & Updates or the Terminal. Both for `ppa:openrazer/stable` and `ppa:polychromatic/edge` (if you'd like to keep getting development updates for that app)

Thanks for checking the command. I'll investigate and fix Polychromatic's troubleshooter. I'll add another check for the latest version.

ScottMorris, 2021-04-21:
Yay 🥳

Thanks so much for you help! I thought I'd reenabled the PPA for _openrazer_ but apparently I didn't.  Enabling that and installing a newer version fixed the issue!

My Huntsman now shows up in the logs,

```
2021-04-21 18:52:53 | razer                          | INFO     | Found valid device.0: 0003:1532:0227.000C
2021-04-21 18:52:53 | razer.device0                  | INFO     | Initialising device.0 RazerHuntsman
2021-04-21 18:52:53 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
```

As well as in Polychromatic (although I haven't reverted the `status` change yet.)

[image]

Cheers mate!

closed 2021-04-21 by ScottMorris