openrazer/openrazer #1499: Battery warnings even though the mouse is off

WhyNotHugo, 2021-04-21:
### Distribution

ArchLinux

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm getting rather frequent notifications saying my mouse battery is low and that I should recharge it. However, my mouse is turned off, and stored in a closet in a different room.

Notifications should stop if the mouse is off.

Logs for `openrazer-daemon`:

```
-- Journal begins at Fri 2020-10-09 09:51:37 CEST, ends at Wed 2021-04-21 12:58:42 CEST. --
Apr 20 11:34:28 victory systemd[763]: Starting Daemon to manage razer devices in userspace...
Apr 20 11:34:29 victory openrazer-daemon[773]: Starting daemon.
Apr 20 11:34:29 victory systemd[763]: Started Daemon to manage razer devices in userspace.
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 773
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlade15Advanced2020 ---------------- (1532:0253)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
Apr 20 11:34:29 victory openrazer-daemon[773]: 2021-04-20 11:34:29 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Base ---------------------- (1532:0246)
[... cut ...]
Apr 21 12:48:41 victory openrazer-daemon[773]: 2021-04-21 12:48:41 | razer.device0                  | DEBUG    | DBus call get_battery
Apr 21 12:48:41 victory openrazer-daemon[773]: 2021-04-21 12:48:41 | razer.device0.batterynotifier  | DEBUG    | Razer Lancehead Wireless Battery at 0.0%
Apr 21 12:58:42 victory openrazer-daemon[773]: 2021-04-21 12:58:42 | razer.device0                  | DEBUG    | DBus call get_battery
Apr 21 12:58:42 victory openrazer-daemon[773]: 2021-04-21 12:58:42 | razer.device0.batterynotifier  | DEBUG    | Razer Lancehead Wireless Battery at 0.0%
```

nightsky30, 2021-04-21:
The mouse may be turned off and stored in another room, but the receiver is its own device which reads the status from the mouse (when available). Not sure there is a way for the receiver to determine if a mouse is just off or if the battery is dead. No response from a battery powered mouse could indicate both possibilities.

Based on those logs, the receiver is still inserted in your system. Apart from changes to how the daemon reads battery status (as in how often or when it polls), the workaround for this issue would be to remove the receiver that isn't being used.

WhyNotHugo, 2021-04-21:
Can I somehow configure it to only report a dead mouse just once?

I do want notifications for _low_ battery though (when battery is > 0).

nightsky30, 2021-04-21:
There are checks which appear to be logging different output based on battery level, but nothing that seems to shutdown the notifier/logging when it gets that low, no:

https://github.com/openrazer/openrazer/blob/89b2e5808bd20b99ccb871957c4b0ffc51f612a6/daemon/openrazer_daemon/misc/battery_notifier.py#L82-L89

PR #1509 (2021-06-20): Fix battery notifier log spamming when mouse is dead or off
Fix battery notifier log spamming when mouse is dead or off
- Fixes #1499
closed 2021-06-20 by z3ntu