openrazer/openrazer #1500: Manjaro Support

littlek3000, 2021-04-22:
Supporting Manjaro would be great if possible, thanks!

z3ntu, 2021-04-22:
Just install it from the AUR? (See Arch Linux instructions on the website)

z3ntu, 2021-04-22:
Try entering "how to build package from aur" into Google?

tildearrow, 2021-04-22:
Here are some basic instructions to save you the time:

0. Make sure you have installed `base-devel`. If not, then install it (open a terminal and type `sudo pacman -S base-devel`).
1. Open a terminal
2. Type `git clone https://aur.archlinux.org/openrazer.git` and press Enter
3. Type `cd openrazer` and press Enter
4. Type `makepkg -sri` and press Enter
5. Wait for it
6. Congratulations, OpenRazer should be installed in your system.
7. Reboot to finish the installation.

autotechmusic, 2021-04-23:
I have v3.0.1 installed on Manjaro KDE.  I didn't have any trouble with either Pamac or Octopi (after adding AUR).  Only trouble I'm having is getting Razergenie to remember my custom color layout after reboot.  But that might be something else for another day.

AT
Manjaro 21.0.2
KDE Plasma

kireita, 2021-04-24:
Hi,

I'm running manjaro as well but i have this problem:
`==> Unable to install module openrazer-driver/3.0.1 for kernel 5.10.30-1-MANJARO: Missing kernel headers.
==> Unable to install module openrazer-driver/3.0.1 for kernel 5.11.14-1-MANJARO: Missing kernel headers.
==> Unable to install module openrazer-driver/3.0.1 for kernel 4.14.230-1-MANJARO: Missing kernel modules tree.
`
how can i solve this?

lah7, 2021-04-24:
@kireita Install `linux-headers` and then the driver:

    sudo pacman -S linux-headers
    sudo dkms install openrazer-driver/3.0.1

Then when the kernel updates in future, DKMS will automatically rebuild the driver.

---

The custom layout across reboots mentioned by @autotechmusic can be solved by setting `restore_persistence = False` in `~/.config/openrazer/razer.conf`. This mainly works for older hardware. (See #1484)

---

Regarding AUR, I find it's easier to use an AUR helper like `yay`:

    sudo pacman -S yay
    yay -S openrazer-meta

kireita, 2021-04-24:
> @kireita Install `linux-headers` and then the driver:
>
> ```
> sudo pacman -S linux-headers
> sudo dkms install openrazer-driver/3.0.1
> ```
>
> Then when the kernel updates in future, DKMS will automatically [...]

i installed the headers and now i get this:
`Error! echo
Your kernel headers for kernel 5.11.14-1-MANJARO cannot be found at
/usr/lib/modules/5.11.14-1-MANJARO/build or /usr/lib/modules/5.11.14-1-MANJARO/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
`

kireita, 2021-04-24:
> i installed the headers and now i get this:
> `Error! echo Your kernel headers for kernel 5.11.14-1-MANJARO cannot be found at /usr/lib/modules/5.11.14-1-MANJARO/build or [...]

solved it by installing the headers of my kernel, in my case it was  `sudo pacman -S linux511-headers` by default it was installing another one, now it detects my hardware correctly

Pandaroses, 2021-05-01:
> solved it by installing the headers of my kernel, in my case it was `sudo pacman -S linux511-headers` by default it was installing another one, now it detects my hardware correctly

after i followed all instructions it said this `Unable to install module openrazer-driver/3.0.1 for kernel 5.11.16-2-MANJARO: Missing kernel modules tree.
`

closed 2021-09-30 by github-actions[bot]