openrazer/openrazer #1503: (Question) Is Razer's Per Key RGB thing lie?

LuxuryWasTaken, 2021-04-28:
Hey, I'm currently using Razer Blade Base 2020 with macOS (They ported openrazer driver to Mac)

And saw that Ripple effect is working for my base model. Razer macOS currently is not working with Wave so I can't say it's per key RGB, but just wondering, is per key rgb / zone rgb a lie?

Is this software limitation for Razer Synapse?

Couldn't boot a linux, so couldn't test it in linux.

lah7, 2021-04-28:
According to OpenRazer's daemon code, the Razer Blade Base Early 2020 has a matrix dimension of `[1, 16]` (1 row, 16 columns). Other Blade models use `[6, 16]` which covers each key on the keyboard.

https://github.com/openrazer/openrazer/blob/89b2e5808bd20b99ccb871957c4b0ffc51f612a6/daemon/openrazer_daemon/hardware/keyboards.py#L1177-L1193

So... it is possible that Razer simplified the keyboard lighting in this model, or the dimensions contributed here could be incorrect - the original issues (#1171, #1150) haven't highlighted any issues.

To know for sure, boot [Ubuntu](https://ubuntu.com) (or other Linux) off a USB drive and test/hack OpenRazer. This project doesn't currently support macOS.

LuxuryWasTaken, 2021-04-28:
It says restart your computer..
(Linux Mint)