openrazer/openrazer #1506: Support for Razer Thunderbolt 4 Dock Chroma

SylvainRamousse, 2021-05-02:
### Device Info

* Name: Thunderbolt 4 Dock Chroma
* Product Number: RC21-0169
* Model Number: RC21-01690100

### Device Information
```
Sorry, I'm a mac user, can't run theses commands
```

j-maynard, 2021-06-14:
I can shed more light on this on the Linux side.  The Thunderbolt docks entry under lspci is:

```
07:00.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 02)
08:00.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 02)
08:01.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 02)
08:02.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 02)
08:03.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 02)
08:04.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 02)
09:00.0 USB controller: Intel Corporation Thunderbolt 4 USB Controller [Goshen Ridge 2020] (rev 02)
```

The lsusb is as follows:
```
Bus 005 Device 013: ID 1532:122e Razer USA, Ltd Thunderbolt 4 Dock
Bus 005 Device 010: ID 1532:0f21 Razer USA, Ltd RAZER THUNDERBOLT 4 DOCK CHROMA
```

Mah thinking is the Chroma control will sit on the USB so its just like any other Razer USB device.  This still needs a pcap which I could do with some simple instructions on how to get for you.

z3ntu, 2021-06-20:
@j-maynard Getting pcaps is documented at https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

In general I don't imagine there to be much useful information in the pcaps, as is with all newish devices - it's probably going to be a case of trial-and-error on the person who implements support. But given that the device also exposes a USB interface I don't expect there to be major differences to the other devices that are supported.

j-maynard, 2021-07-06:
Synapse Pictures:
[image]
[image]

j-maynard, 2021-07-17:
Sorry I skipped the most important screen shot (The one which shows how many LEDs there are to control in Chroma Studio).  

Now I've done the PR for the Blade Advanced (https://github.com/openrazer/openrazer/pull/1561) I think I might try looking at this to see if I can make it work.  If anyone with experience of the code base has ideas I am happy to hear them.

[image]

PR #1635 (2021-10-30): Add Razer Thunderbolt 4 Dock Chroma Support
Adds Razer Thunderbolt 4 Dock Chroma Support

Fixes #1506

Works:
 - [x] Breath Single
 - [x] Breath Dual
 - [x] Breath Random
 - [x] Starlight Single
 - [x] Starlight Dual
 - [x] Starlight Random
 - [x] Wave
 - [x] None
 - [x] Spectrum
 - [x] Static
 - [x] Brightness

### Device Model Info

* Name: Razer Thunderbolt 4 Dock Chroma
* Product Number: RC21-01690100
* Model Number: RC21-0169

### Device Information

<details>
<summary>
ls -lah /dev/input/by-id/
</summary>

N/A

</details>
<details>
<summary>
lsusb -d 1532: -v
</summary>

Bus 004 Device 0 [... cut]
closed 2021-10-30 by z3ntu