openrazer/openrazer #1507: Support for [device name here]

Astilex5, 2021-05-04:
### Device Info

* Name: Razer Naga Classic Edition
* Product Number: RZ01-02410200
* Model Number: RZ01-0241

### Device Information
```
Please post the output of
    `Bus 003 Device 003: ID 1532:0511 Razer USA, Ltd RZ19-0229 Gaming Microphone
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        16
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0511 RZ19-0229 Gaming Microphone
  bcdDevice            1.00
  iManufacturer           1 Razer Inc
  iProduct                2 Razer Seiren X
  iSerial                 3 UC2046L01305036
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0121
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x006a
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                12
[... 930 lines cut ...]
```
Please post the output of
    `total 0
drwxr-xr-x 2 root root 280 May  3 20:55 .
drwxr-xr-x 4 root root 720 May  3 20:55 ..
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-046d_0825_C2A9A5A0-event-if00 -> ../event1
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-Razer_Inc_Razer_Seiren_X_UC2046L01305036-event-if03 -> ../event8
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-Razer_Razer_Huntsman_Elite-event-kbd -> ../event0
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-Razer_Razer_Huntsman_Elite-if01-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-Razer_Razer_Huntsman_Elite-if02-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-Razer_Razer_Huntsman_Elite-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 May  3 20:55 usb-Razer_Razer_Naga_Classic_Edition_00000000001A-event-if01 -> ../event12
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-Razer_Razer_Naga_Classic_Edition_00000000001A-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 May  3 20:55 usb-Razer_Razer_Naga_Classic_Edition_00000000001A-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 May  3 20:55 usb-Razer_Razer_Naga_Classic_Edition_00000000001A-if02-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-Razer_Razer_Naga_Classic_Edition_00000000001A-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 May  3 20:55 usb-Razer_Razer_USB_Audio_Controller_000000000000000000000000-event-if05 -> ../event7
`
```

### Packet Captures

### Synapse Screenshots

NOTE: This is a new Model I found at Walmart for $70, It looks like a Naga Trinity minus the swapable buttons. Sadly it is not listed on Razer's support site. So I am guessing it was released in 2021.

TheFALSEGamer4, 2021-06-21:
The device ID for this mouse is 1532:0093 btw. I didn't see it in the issue so I thought that I would add it in case you guys needed it.

boddahburnout, 2021-09-06:
In the same boat here with the same Pid also on latest openrazor, was hoping I could remap my plate keys

z3ntu, 2021-10-01:
@boddahburnout For key remapping you can use https://github.com/sezanzeb/key-mapper without openrazer support.

zacharygriffee, 2023-05-22:
I am in the same boat with this device. Input Mapper doesn't recognize my 7-12 "plate keys" buttons to remap. I use these buttons very extensively both in programming and game playing before I swapped to Arch.