openrazer/openrazer #1508: Question about storing the setting on devices without onboard memory

wwwkris39, 2021-05-05:
Sorry if it's not a right place to ask but I don't see any other way.
How does storing settings on devices without onboard memory work? I have Abyssus Essential mouse and when I set the lightning and DPI settings in synapse and then uninstall it, it reverts to defaults but when I changed the settings on your software, they stayed even when I connected the mouse to a different laptop.
I'm just asking because I'm very curious since supposedly it doesn't have any memory

lah7, 2021-05-05:
OpenRazer communicates with the firmware directly, so it sets device settings at a hardware level. Razer Synapse on the other hand performs _mostly everything_ in software, even the rendering of effects.

* **Razer Synapse 3** → Constantly sends new frames over USB 🖌️ 
* **OpenRazer / Synapse 2** → "Hey, set your effect to wave" 👉🏼 

It seems that many of the _older_ hardware are able to retain their settings across reboots and when unplugged, but some of the newer hardware always reset to a default (e.g. Spectrum) and relies on software (OpenRazer's persistence, or Synapse 3) to restore their previous settings. Some firmware may also reset to a default at power up if it was previously rendering a custom frame via software.

I'm not sure on the broader "onboard memory", as some mice apparently can store DPI settings.

wwwkris39, 2021-05-05:
I see. So, basically, on older. cheaper models they just force the user to use their software probably because they want to artificially raise the prices of the mice with their own memory, even though they probably use some of the same parts since it is possible to store settings on them. Thank you for your answer!

closed 2021-05-05 by wwwkris39

lah7, 2021-05-06:
> OpenRazer communicates with the firmware directly, so it sets device settings at a hardware level.

Just to clarify that - OpenRazer is actually the one instructing the hardware to save the effect - which is the `VARSTORE` thing - for example, when setting a wave effect:

https://github.com/openrazer/openrazer/blob/89b2e5808bd20b99ccb871957c4b0ffc51f612a6/driver/razeraccessory_driver.c#L595

https://github.com/openrazer/openrazer/blob/89b2e5808bd20b99ccb871957c4b0ffc51f612a6/driver/razercommon.h#L40-L42

----

I would **guess** Razer went software only with Synapse 3 to offer more effect customisation/flexibility, at the expense of the driver/CPU having to do all the work. I have a Razer Base Station Chroma (headphone stand) which is a "Synapse 3 only" device, but sure enough we discovered it still supports hardware effects.