openrazer/openrazer #1512: DeathAdder 1800 stuttering and DPI issue Manjaro

Kern3l-Panic, 2021-05-13:
### Distribution

### Support

- [Yes] Have you tried rebooting?
- [Yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

My DeathAdder 1800 stutters when I move it too fast. It works fine at lower speeds, but stutters in the middle of moving too fast. I have a Razer Firefly mouse mat so I know it's not a surface issue. I've used compressed air to clear the PC USB ports and USB connector of the mouse. I've tried different polling rates: 125Hz-1000Hz. I've also reinstalled openrazer-meta and polychromatic from the AUR. Lastly, I tested the mouse on two different Linux Mint PCs, and both display NO signs of stuttering. One had the same Razer drivers and the other had none installed.

I'm unsure if it's related but the UI displays incorrect DPI values as seen in the attachment. Another problem is when I increase the polling rate I hear buzzing/hissing sounds in my headphones.
[image]

lah7, 2021-05-13:
For the UI issue, I'd recommend installing `polychromatic-git` (v0.6.0) instead of `polychromatic` (v0.3.12) from the AUR and see if the problem persists.

If the hardware is only problematic under this Manjaro PC, check the kernel log to see if the driver reports any errors:

    sudo dmesg | tail

You could try the `linux-lts`/`linux-lts-headers` (5.10.36) package instead of `linux`/`linux-headers` (5.12.3) in case it's a problem with the newer, more cutting edge kernel. I imagine Linux Mint uses LTS kernels.

Unfortunately, the buzzing/hissing via headphones (presuming via 3.5mm AUX connection) could be signs of a grounding issue with the motherboard. Alternately, since [higher polling rate = CPU polls mouse more frequently](https://www.howtogeek.com/182702/mouse-dpi-and-polling-rates-explained-do-they-matter-for-gaming/) there might be a bottleneck with the USB bus which might cause the stutter. Just a guess, I'm not 100% sure on the hardware side of things.

Kern3l-Panic, 2021-05-14:
> For the UI issue, I'd recommend installing `polychromatic-git` (v0.6.0) instead of `polychromatic` (v0.3.12) from the AUR and see if the problem persists.
>
> If the hardware is only problematic under [...]

I removed the older version of polychromatic and installed the newer one. Unfortunately, the UI still reports incorrect DPI values as seen in the attachment. It reports a very high DPI X value and a DPI Y value of 0.

For the hardware problem, it's NOT Arch/Manjaro related and has nothing to do with the Kernel. Since I used different PCs I also used different mouse mats. My Razer Firefly is actually the problem which I didn't think could be possible. Using my desk or a blank sheet of paper fixes the stuttering, but they're uncomfortable and inferior. I hope a calibration tool gets implemented soon so the stuttering can be resolved.

For the buzzing/hissing from my headphones, it's only noticeable when increasing the polling rate to 1000Hz. The solution is to use 500Hz which works fine.

[image]

lah7, 2021-05-15:
Quick note - in this current state of Polychromatic shown here, you'll need to press <kbd>F5</kbd> (Refresh) to see the new number under the title, which will fetch it from OpenRazer's daemon again. 

It seems the DeathAdder 1800 uses a slightly different method (`get_dpi_xy_byte`) then other mice like the Razer Mamba:

https://github.com/openrazer/openrazer/blob/89b2e5808bd20b99ccb871957c4b0ffc51f612a6/daemon/openrazer_daemon/hardware/mouse.py#L1981-L1990

Looking at the original pull request, it seems it was "normal" that the initial value was bogus: https://github.com/openrazer/openrazer/pull/939#issuecomment-546637040. To explain the UI behaviour, the old Polychromatic asks the daemon for the value again when changed (hence, weird jumping numbers), but the new one (and RazerGenie) does not do that. In any case they are being fed a bad value initially.

To bypass frontends, you can interact with the OpenRazer API directly:

Set DPI X/Y:

    python3 -c "import openrazer.client; openrazer.client.DeviceManager().devices[0].dpi = (1000, 1000)"
    python3 -c "import openrazer.client; openrazer.client.DeviceManager().devices[0].dpi = (1000, 0)"
    python3 -c "import openrazer.client; openrazer.client.DeviceManager().devices[0].dpi = (1000, -1)"

Get DPI:

    python3 -c "import openrazer.client; print(openrazer.client.DeviceManager().devices[0].dpi)"

(Change `devices[0]` to `devices[1]` if the first one is your Firefly)

To read the sysfs (driver) filesystem directly - the path will be similar to this:

    cat /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/0003:1532:0046.0001/dpi

It will require someone with this mouse to investigate and fix or understand the "get" DPI for this device. Mathematically do these values have a trend. or are they always random?

Kern3l-Panic, 2021-05-16:
> Quick note - in this current state of Polychromatic shown here, you'll need to press F5 (Refresh) to see the new number under the title, which will fetch it from OpenRazer's daemon again.
> 
> It seems the DeathAdder 1800 uses a slightly different method (`get_dpi_xy_byte`) then other mice like the Razer Mamba:
> 
> https://github.com/openrazer/openrazer/blob/89b2e5808bd20b99ccb871957c4b0ffc51f612a6/daemon/openrazer_daemon/hardware/mouse.py#L1981-L1990
> 
> Looking at the original pull request, it seems it was "normal" that the initial value was bogus: [#939 (comment)](https://github.com/openrazer/openrazer/pull/939#issuecomment-546637040). To explain the UI behaviour, the old Polychromatic asks the daemon for the value again when changed (hence, weird jumping numbers), but the new one (and RazerGenie) does not do that. In any case they are being fed a bad value initially.
> 
> To bypass frontends, you can interact with the OpenRazer API directly:
> 
> Set DPI X/Y:
> 
> ```
> python3 -c "import openrazer.client; openrazer.client.DeviceManager().devices[0].dpi = (1000, 1000)"
> python3 -c "import openrazer.client; openrazer.client.DeviceManager().devices[0].dpi = (1000, 0)"
> python3 -c "import openrazer.client; openrazer.client.DeviceManager().devices[0].dpi = (1000, -1)"
> ```
> 
> Get DPI:
> 
> ```
> python3 -c "import openrazer.client; print(openrazer.client.DeviceManager().devices[0].dpi)"
> ```
> 
> (Change `devices[0]` to `devices[1]` if the first one is your Firefly)
> 
> To read the sysfs (driver) filesystem directly - the path will be similar to this:
> 
> ```
> cat /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/0003:1532:0046.0001/dpi
> ```
> 
> It will require someone with this mouse to investigate and fix or understand the "get" DPI for this device. Mathematically do these values have a trend. or are they always random?

The values are not random. A very high value consistently represents the normal DPI values: 250279 DPI = 1000 DPI