openrazer/openrazer #1513: file is not owned by plugdev

gprime31, 2021-05-14:
dist: Arch 64

2021-05-14 08:44:10 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.2/0003:1532:011A.0007/device_type, file is not owned by plugdev
2021-05-14 08:44:10 | razer                          | INFO     | Serving DBus

from :  
~/.local/share/openrazer/logs/razer.log

and on Polychromatic it says device can be accessed using plugdev permissions and had an x
fixed it bu changing the group on the device to plugdev

sudo chown root:plugdev /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.2/0003:1532:011A.0007/device_type
for anyone who has this issue.

gprime31, 2021-05-14:
[image]

gprime31, 2021-05-14:
if i change the owner to root:plugdev it fixes the permission problem
but then i get:
permissionError: [Errno 13] Permission denied: '/dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate-if01-event-kbd'
when running openrazer-daemon -Fv

gprime31, 2021-05-14:
ok then do:
sudo chown $user:plugdev /dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate-event-kbd
and
sudo chown $user:plugdev /dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate-if01-event-kbd
and everything passes checks.
[image]

gprime31, 2021-05-14:
and works
But, i have to do this every boot.
any suggestions??

lah7, 2021-05-15:
It sounds like [udev rules](https://wiki.archlinux.org/title/Udev) are not working on your Arch install. I'm not sure exactly how to fix this, but installing `openrazer-meta` from AUR should have installed the file `/usr/lib/udev/rules.d/99-razer.rules` required to automatically set the permissions.

gprime31, 2021-05-15:
yes, it missing.
find: ‘99-razer.rules’: No such file or directory
where can i find it?

gprime31, 2021-05-15:
seems like openrazer/install_files/udev/99-razer.rules is missing from the latest release. i found in on an older build:
https://github.com/openrazer/openrazer/blob/master/install_files/udev/99-razer.rules
will try this.

gprime31, 2021-05-15:
that fixed it ^^^^ works on boot now, thanks.

lah7, 2021-05-15:
The file is provided in `openrazer-driver-dkms` (AUR) [corrected above, `openrazer-meta` was wrong, my bad]. 

Glad you found a solution! For some reason, your system had misplaced the files. Hopefully this won't happen again when you next update the packages.

closed 2021-05-15 by lah7

gprime31, 2021-05-16:
[image]

gprime31, 2021-05-16:
So it's the wrong package listed on the website? ^^^^

lah7, 2021-05-16:
`openrazer-meta` is correct, it's [a meta package](https://wiki.archlinux.org/title/Meta_package_and_package_group) that depends on `openrazer-daemon`, `openrazer-driver-dkms` and `python-openrazer`. Installing that depends on these 3 packages so you can easily install using an AUR helper like `yay`:

    yay -S openrazer-meta

My clarification above was that the udev rule file you were missing is provided in `openrazer-driver-dkms`, so that would have been the package to reinstall, as reinstalling `openrazer-meta` would not have worked.

gprime31, 2021-05-16:
Ok, I see, thanks.

xamindar, 2023-10-19:
This is still an issue with the latest arch packages. I have tried both the standard package and the aur ones and none of them install the udev rule for permissions.