openrazer/openrazer #1517: Razer Viper (1532:0078) not fully supported?

ychromosome, 2021-05-21:
### Distribution

Fedora F34

### Support

- [ ] Have you tried rebooting?
- [ ] yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?
- [ ] yes

I installed openrazer according to the guides on the website. 
Razergenie detects the device, I can change the DPI and the illumination of the mouse logo, but when I want to open the "Custom Editor" I get told, that I should file a report for "mouse".
Razergenie mentions I'm running Daemon 3.0.1

Bus 001 Device 003: ID 1532:0078 Razer USA, Ltd Viper (wired)

dmesg | grep -i razer
[    2.271479] usb 1-5: Product: Razer Viper
[    2.271480] usb 1-5: Manufacturer: Razer
[    2.292497] input: Razer Razer Viper as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-5/1-5:1.0/0003:1532:0078.0003/input/input5
[    2.292585] hid-generic 0003:1532:0078.0003: input,hidraw2: USB HID v1.00 Mouse [Razer Razer Viper] on usb-0000:02:00.0-5/input0
[    2.299553] input: Razer Razer Viper Keyboard as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-5/1-5:1.1/0003:1532:0078.0004/input/input6
[    2.351253] input: Razer Razer Viper as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-5/1-5:1.1/0003:1532:0078.0004/input/input7
[    2.351296] hid-generic 0003:1532:0078.0004: input,hidraw3: USB HID v1.00 Keyboard [Razer Razer Viper] on usb-0000:02:00.0-5/input1
[    2.355500] input: Razer Razer Viper as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-5/1-5:1.2/0003:1532:0078.0005/input/input8
[    2.407359] hid-generic 0003:1532:0078.0005: input,hidraw4: USB HID v1.00 Keyboard [Razer Razer Viper] on usb-0000:02:00.0-5/input2
[   16.462621] razermouse: loading out-of-tree module taints kernel.
[   16.487228] razermouse: module verification failed: signature and/or required key missing - tainting kernel
[   16.547225] input: Razer Razer Viper as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-5/1-5:1.0/0003:1532:0078.0003/input/input12
[   16.547307] razermouse 0003:1532:0078.0003: input,hidraw2: USB HID v1.00 Mouse [Razer Razer Viper] on usb-0000:02:00.0-5/input0
[   16.609188] input: Razer Razer Viper as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-5/1-5:1.1/0003:1532:0078.0004/input/input13
[   16.661075] razermouse 0003:1532:0078.0004: input,hidraw3: USB HID v1.00 Keyboard [Razer Razer Viper] on usb-0000:02:00.0-5/input1
[   16.702214] input: Razer Razer Viper as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-5/1-5:1.2/0003:1532:0078.0005/input/input27
[   16.758114] razermouse 0003:1532:0078.0005: input,hidraw4: USB HID v1.00 Keyboard [Razer Razer Viper] on usb-0000:02:00.0-5/input2

journalctl -b | grep -i razer_mount
Mai 21 17:30:52 fedora razer_mount[1266]: Driver razermouse
Mai 21 17:30:52 fedora razer_mount[1275]: Device_ID 1-5
Mai 21 17:30:52 fedora razer_mount[1276]: Modprobing razermouse
Mai 21 17:30:53 fedora razer_mount[1371]: Modprobed razermouse
Mai 21 17:30:53 fedora razer_mount[1379]: Driver razermouse
Mai 21 17:30:53 fedora razer_mount[1381]: Driver razermouse
Mai 21 17:30:53 fedora razer_mount[1380]: Driver razermouse
Mai 21 17:30:53 fedora razer_mount[1382]: Device_ID 1-5:1.1
Mai 21 17:30:53 fedora razer_mount[1383]: Device_ID 1-5:1.2
Mai 21 17:30:53 fedora razer_mount[1384]: Device_ID 1-5:1.0
Mai 21 17:30:53 fedora razer_mount[1388]: Driver razermouse
Mai 21 17:30:53 fedora razer_mount[1390]: Driver razermouse
Mai 21 17:30:53 fedora razer_mount[1389]: Driver razermouse
Mai 21 17:30:53 fedora razer_mount[1391]: Device_ID 0003:1532:0078.0004
Mai 21 17:30:53 fedora razer_mount[1392]: Device_ID 0003:1532:0078.0005
Mai 21 17:30:53 fedora razer_mount[1393]: Device_ID 0003:1532:0078.0003
Mai 21 17:30:53 fedora razer_mount[1394]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0004/
Mai 21 17:30:53 fedora razer_mount[1395]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0005/
Mai 21 17:30:53 fedora razer_mount[1396]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0003/
Mai 21 17:30:53 fedora razer_mount[1400]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0004/
Mai 21 17:30:53 fedora razer_mount[1401]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0003/
Mai 21 17:30:53 fedora razer_mount[1402]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0005/

2021-05-21 17:13:51 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 4983
2021-05-21 17:13:51 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-05-21 17:13:51 | razer                          | INFO     | Found device.0: 0003:1532:0078.0003
2021-05-21 17:13:51 | razer.device0                  | INFO     | Initialising device.0 RazerViper
2021-05-21 17:13:51 | razer                          | INFO     | Serving DBus
2021-05-21 17:21:00 | razer                          | INFO     | Stopping daemon on signal 15
2021-05-21 17:33:22 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 5423
2021-05-21 17:33:22 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-05-21 17:33:22 | razer                          | INFO     | Found device.0: 0003:1532:0078.0003
2021-05-21 17:33:22 | razer.device0                  | INFO     | Initialising device.0 RazerViper
2021-05-21 17:33:22 | razer                          | INFO     | Serving DBus
/home/martin/.local/share/openrazer/logs/razer.log (END)

from my side it looks right, maybe you can give me a hand to configure the buttons properly.

nightsky30, 2021-05-21:
The custom editor hasn't been functional for mice for a while now. It's not this mouse in particular. :)

ychromosome, 2021-05-21:
oh ok, sorry for the question then.
How do I change the Button Layout then?

nightsky30, 2021-05-21:
I don't believe the custom editor in RazerGenie has ever handled anything as far as button or key layout/binding if that's what you are referring to. The custom editor only handles setting custom LED colors (the current behavior of the custom editor with keyboards anyway). Glanced at the code, and it doesn't seem to indicate otherwise.

There were some PRs related to button/key re-binding for the daemon. Not sure what their state is.

closed 2021-10-19 by github-actions[bot]