openrazer/openrazer #1519: error: ‘struct hid_driver’ has no member named ‘match’

eadmaster, 2021-05-23:
### Distribution

Ubuntu 18.04.2 LTS

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

building the latest dkms fails in my ubuntu machine, while the prev version was working fine:
````
cat /var/lib/dkms/openrazer-driver/3.0.1/build/make.log

DKMS make.log for openrazer-driver-3.0.1 for kernel 4.15.0-50-generic (x86_64)
Sun May 23 13:51:16 CEST 2021
-e 
:: Compiling OpenRazer kernel modules
========================================
make -C /lib/modules/4.15.0-50-generic/build M=/var/lib/dkms/openrazer-driver/3.0.1/build/driver modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-50-generic'
Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razerkbd_driver.o
  CC [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razercommon.o
  CC [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razerchromacommon.o
  LD [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razerkbd.o
  CC [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razermouse_driver.o
  LD [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razermouse.o
  CC [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razerkraken_driver.o
  LD [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razerkraken.o
  CC [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razeraccessory_driver.o
/var/lib/dkms/openrazer-driver/3.0.1/build/driver/razeraccessory_driver.c:2019:6: error: ‘struct hid_driver’ has no member named ‘match’
     .match = razer_accessory_match,
      ^~~~~
/var/lib/dkms/openrazer-driver/3.0.1/build/driver/razeraccessory_driver.c:2019:14: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
     .match = razer_accessory_match,
              ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/openrazer-driver/3.0.1/build/driver/razeraccessory_driver.c:2019:14: note: (near initialization for ‘razer_accessory_driver.dyn_list.next’)
/var/lib/dkms/openrazer-driver/3.0.1/build/driver/razeraccessory_driver.c:2016:51: warning: missing braces around initializer [-Wmissing-braces]
 static struct hid_driver razer_accessory_driver = {
                                                   ^
/var/lib/dkms/openrazer-driver/3.0.1/build/driver/razeraccessory_driver.c:2019:14:
     .match = razer_accessory_match,
              {                    }
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:332: /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razeraccessory_driver.o] Error 1
make[1]: *** [Makefile:1552: _module_/var/lib/dkms/openrazer-driver/3.0.1/build/driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-50-generic'
make: *** [Makefile:25: driver] Error 2
````

lah7, 2021-05-23:
I just tested an up-to-date Ubuntu MATE 18.04 (HWE kernel 5.4.0) and DKMS built with no errors. According to your output, you may be missing some packages:

> Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"

Try installing them, or `build-essential`:

    sudo apt install build-essential

eadmaster, 2021-05-26:
> Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"

I've installed `libelf-dev`, and that warning disappeared, but it still fails with the same error.

z3ntu, 2021-05-26:
It looks like since v3.0.0 we do not support kernel 4.15 anymore, just 4.16+. Please try upgrading to the HWE kernel as mentioned in the last paragraph of the last comment.

eadmaster, 2021-05-26:
ok ,now the error is different: :-)
````
DKMS make.log for openrazer-driver-3.0.1 for kernel 5.4.0-73-generic (x86_64)
Wed May 26 15:03:55 CEST 2021
-e 
:: Compiling OpenRazer kernel modules
========================================
make -C /lib/modules/5.4.0-73-generic/build M=/var/lib/dkms/openrazer-driver/3.0.1/build/driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-73-generic'
  CC [M]  /var/lib/dkms/openrazer-driver/3.0.1/build/driver/razerkbd_driver.o
In file included from <command-line>:0:0:
./arch/x86/include/asm/page_64.h: In function ‘clear_page’:
././include/linux/compiler_types.h:214:24: error: expected ‘(’ before ‘__inline’
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/alternative.h:256:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile (ALTERNATIVE_2("call %P[old]", "call %P[new1]", feature1,\
  ^~~~~~~~~~
... (full log attached)
````

[make.log](https://github.com/openrazer/openrazer/files/6546875/make.log)

eadmaster, 2021-05-26:
the default gcc version is 7.4.0 in ubuntu1~18.04.
Maybe this can be fixed by switching to the latest 8.4.0?

z3ntu, 2021-05-26:
Another alternative is upgrading to Ubuntu 20.04

eadmaster, 2021-05-26:
of course 😉

lah7, 2021-05-26:
You mention your distribution is 18.04.2, but the latest should be 18.04.5:

    lsb_release -a

Could your system be missing updates or packages for compiling software? For sure, on an empty 18.04.5 and kernel `5.4.0-73-generic` (no changes to `gcc`), I can add `ppa:openrazer/stable` and install the DKMS module successfully without any errors. Even `4.15.0-143-generic` is successful too.

eadmaster, 2021-05-26:
Yes, then i guess it must be some outdated package: when you install from scratch you automatically grab the latest version of each package...