openrazer/openrazer #1520: Razer Kraken Kitty Edition Missing Chroma options

Kusoneko, 2021-05-29:
### Distribution

Gentoo
Openrazer 3.0.1

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Sister issue in Polychromatic here: https://github.com/polychromatic/polychromatic/issues/328#issue-906327255

Hello,

I have a Razer Kraken Kitty Edition Headset. According to the Razer website, and on the Synapse 3 application on Windows, it supports the following Chroma schemes:

- Breath
- Fire
- Spectrum
- Starlight
- Static
- Wave
- Wheel

However, on openrazer (after looking into where the Kitty edition is set up in the accessory driver), there is currently only the following options:

 - None
 - Static
 - Breath

I would like to use Spectrum and have it synced with the rest of my Razer devices. I tried to "force" it using polychromatic-cli by doing the following command: `polychromatic-cli -d headset -o spectrum` however, that gives me the following error message: `Skipping: Razer Kraken Kitty Edition - Setting this option was unsuccessful or is not supported.`

Here are my various version information strings:
OS: Gentoo
Openrazer: 3.0.1 (rebuilt today just in case)
Polychromatic: 0.6.0 (rebuilt today just in case)

Is there any way to enable chroma options that are disabled for some reason on polychromatic despite the device supporting them?

Thanks,
Kusoneko

999eagle, 2021-06-04:
Okay, I'll answer more here. Razer Synapse 3 actually uses software based lighting for everything on the Kraken Kitty Edition. While working on support for that in openrazer I captured packets Synapse sent and all effects (including static lighting) were made through sending new colors every few milliseconds. For the openrazer implementation we don't have any kind of software based lighting yet and can only use hardware based effects, so the three effects in openrazer are hardware effects the Kraken Kitty Edition seems to support, but Synapse actually never uses them.  
I can try again this weekend to see if some other hardware effects are also supported, but as I said in the other issue, if I recall correctly they didn't work during my testing.

999eagle, 2021-06-11:
Hi, it took me a bit longer than I thought, but I got the Starlight, Spectrum and Wave effects working on my headset. Didn't manage to test Fire and Wheel yet, but I've opened a PR adding the three tested and working effects.

Kusoneko, 2021-06-13:
Thank you for your help!

closed 2021-06-20 by z3ntu