openrazer/openrazer #1524: Support for Razer Blade Base (Late 2020)

snowmandev, 2021-06-08:
### Device Info

* Name: Razer Blade 15 (Late 2020)
* Product Number: RZ09-03519E11
* Model Number: RZ09-0351

### Device Information
```
Bus 001 Device 003: ID 1532:0268 Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0268 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
total 0
drwxr-xr-x 2 root root 160 Jun  8 23:03 .
drwxr-xr-x 4 root root 660 Jun  8 23:03 ..
lrwxrwxrwx 1 root root  10 Jun  8 23:03 usb-Chicony_Electronics_Co._Ltd._HD_Webcam_0001-event-if00 -> ../event11
lrwxrwxrwx 1 root root   9 Jun  8 23:03 usb-Razer_Razer_Blade-event-if01 -> ../event8
lrwxrwxrwx 1 root root   9 Jun  8 23:03 usb-Razer_Razer_Blade-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Jun  8 23:03 usb-Razer_Razer_Blade-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root  10 Jun  8 23:03 usb-Razer_Razer_Blade-if02-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 Jun  8 23:03 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
```

### Packet Captures
Will work on this.

### Synapse Screenshots
Will work on this.

turbos-and-tacos, 2021-06-19:
If someone can provide a pcap, I can work on this one.

JuliusHerrmann, 2021-06-25:
Is there an update? I just bought this laptop and would really like to use it fully.

sanyer, 2021-07-10:
Hi @turbos-and-tacos 
What's pcap in this context and how can I get it for you?

turbos-and-tacos, 2021-07-10:
Look at the reverse engineering section of the wiki for this project. It has directions for how to collect a pcap. I am unable to get virtualbox working on my new machine. Something weird with the kernel headers is going on.

cogtoto, 2021-10-03:
Hi, any plan for this device to be supported in the future ? do I need to provide more data ? thanks

IliaNoz, 2022-01-27:
Any update about this?

cogtoto, 2022-01-30:
unfortunately no update.... 
I wish I could use openrazor on my laptop :)

Derminator404, 2023-03-26:
So i got working on recording some pcap and also screenshoots 
[image]
[image]
[image]
[image] 

If needed i can provide more records pcap 
[pcaps.zip](https://github.com/openrazer/openrazer/files/11070980/pcaps.zip)

z3ntu, 2023-04-22:
@Derminator404 Thanks, pcaps look very standard so for implementation you can follow basically any other recent Blade laptop and add support like that. Should be fairly straightforward.

miggyG, 2023-07-09:
Is this still being worked on, I have the same laptop and would like to be able to use openrazer, if needed I can provide any necessary files needed to get support

z3ntu, 2023-07-16:
@miggyG Adding support for this device would be best :wink: See https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

HominidHost, 2024-09-17:
Has anyone had any success in adding support for the Razer Blade 15 Base (Late 2020) yet?

Lainius, 2025-02-20:
> Has anyone had any success in adding support for the Razer Blade 15 Base (Late 2020) yet?

I am currently working on this. I believe that, theoretically, the edits I've made ought to work. However, as an amateur programmer, I'm having trouble navigating the ecosystem around the code itself. I have yet been able to figure out how to actually test the new kernel modules. I've passed every check in Polychromatic except the one which confirms the device is in the stable branch. Maybe some more edits to Polychromatic will fix it. No luck with some of the other frontends and the CLI offering. 

My implementation is available nonetheless. I'm not sure whether or not to make a pull request considering there's no guarantee the thing I've made actually works. I will continue to test things and hopefully I can figure something out in the next 48 hours while I have the free time.

Lainius, 2025-02-20:
> Has anyone had any success in adding support for the Razer Blade 15 Base (Late 2020) yet?

I have confirmed that it does, in fact, work. Polychromatic both recognizes it and is able to manipulate the keyboard's RGB. I will make a pull request. I may need to make changes based on what the reviewer says. I've never made a pull request before so I'm not sure about what to expect.

closed 2025-02-25 by z3ntu