openrazer/openrazer #1525: Basilisk X Hyperspeed recognized but not functional through USB dongle

BrandonHarrisonCode, 2021-06-08:
### Distribution

`Arch Linux 5.12.9-arch1-1`
Open razer version `3.0.1`

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I am using the Basilisk X Hyperspeed in the 2.4GHz mode. When the USB receiver is plugged into the computer itself, the mouse works as expected. However, when the USB reciever is plugged into a USB dongle which is then plugged into the computer, the mouse is recognized without errors but all functionality is disabled. This includes moving the mouse, clicking, etc.

There are no errors in `journalctl`, `~/.local/share/openrazer/logs/razer.logs`, `lsusb` correctly recognizes the mouse with or without the dongle, and `polychromatic` indicates that it is successfully able to connect with the mouse regardless of if the receiver is in the dongle or not. As far as I can tell, the logs are the same after connecting the mouse through the dongle or not through the dongle.

z3ntu, 2021-06-20:
> the mouse is recognized without errors but all functionality is disabled. This includes moving the mouse, clicking, etc.

You're saying that the mouse doesn't work as a mouse here? Can you please also try the same but with openrazer uninstalled? In theory it shouldn't affect functionality

BrandonHarrisonCode, 2021-06-21:
Ha, I suppose that's one way of putting it. Yes, the mouse connects to the computer, is recognized in `lsusb` as a Basilisk X Hyperspeed, but the mouse is completely unresponsive - clicks are not registered and the cursor does not move. I'm away from home right now so I will get back to you as soon as I can in about a week about the results of uninstalling openrazer

BrandonHarrisonCode, 2021-07-05:
I uninstalled openrazer and had the same results. After uninstalling, I left the receiver in the USB hub and rebooted. `lsusb` lists the mouse correctly as a Basilisk X Hyperspeed but the issue of the mouse not working persists. Upon moving the receiver from the USB hub directly into the computer the mouse is able to move and click again. Here are logs from `dmesg` where I place it into the extender first, and then remove it and place it directly into the computer:

```
[  709.307103] usb 5-1.1: new full-speed USB device number 9 using xhci_hcd
[  709.507887] usb 5-1.1: New USB device found, idVendor=1532, idProduct=0083, bcdDevice= 2.00
[  709.507901] usb 5-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  709.507906] usb 5-1.1: Product: Razer Basilisk X HyperSpeed
[  709.507910] usb 5-1.1: Manufacturer: Razer
[  709.518574] input: Razer Razer Basilisk X HyperSpeed as /devices/pci0000:00/0000:00:1c.6/0000:09:00.0/usb5/5-1/5-1.1/5-1.1:1.0/0003:1532:0083.0011/input/input38
[  709.518833] razermouse 0003:1532:0083.0011: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Basilisk X HyperSpeed] on usb-0000:09:00.0-1.1/input0
[  709.519789] input: Razer Razer Basilisk X HyperSpeed as /devices/pci0000:00/0000:00:1c.6/0000:09:00.0/usb5/5-1/5-1.1/5-1.1:1.1/0003:1532:0083.0012/input/input39
[  709.573990] razermouse 0003:1532:0083.0012: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Basilisk X HyperSpeed] on usb-0000:09:00.0-1.1/input1
[  709.574672] input: Razer Razer Basilisk X HyperSpeed as /devices/pci0000:00/0000:00:1c.6/0000:09:00.0/usb5/5-1/5-1.1/5-1.1:1.2/0003:1532:0083.0013/input/input40
[  709.630539] razermouse 0003:1532:0083.0013: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Basilisk X HyperSpeed] on usb-0000:09:00.0-1.1/input2
[  714.430144] audit: type=1101 audit(1625526066.746:140): pid=2650 uid=1000 auid=1000 ses=1 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="brandon" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  714.430335] audit: type=1110 audit(1625526066.746:141): pid=2650 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  714.432281] audit: type=1105 audit(1625526066.746:142): pid=2650 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  714.435163] audit: type=1106 audit(1625526066.749:143): pid=2650 uid=1000 auid=1000 ses=1 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  714.435211] audit: type=1104 audit(1625526066.749:144): pid=2650 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  756.465706] usb 5-1.1: USB disconnect, device number 9
[  756.534108] usbhid 5-1.1:1.0: Razer Device disconnected
[  756.627377] usbhid 5-1.1:1.1: Razer Device disconnected
[  756.785434] usbhid 5-1.1:1.2: Razer Device disconnected
[  759.026804] usb 3-8: new full-speed USB device number 5 using xhci_hcd
[  759.168382] usb 3-8: New USB device found, idVendor=1532, idProduct=0083, bcdDevice= 2.00
[  759.168395] usb 3-8: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  759.168401] usb 3-8: Product: Razer Basilisk X HyperSpeed
[  759.168405] usb 3-8: Manufacturer: Razer
[  759.170751] input: Razer Razer Basilisk X HyperSpeed as /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:1532:0083.0014/input/input41
[  759.170878] razermouse 0003:1532:0083.0014: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Basilisk X HyperSpeed] on usb-0000:00:14.0-8/input0
[  759.171570] input: Razer Razer Basilisk X HyperSpeed as /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.1/0003:1532:0083.0015/input/input42
[  759.227109] razermouse 0003:1532:0083.0015: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Basilisk X HyperSpeed] on usb-0000:00:14.0-8/input1
[  759.228170] input: Razer Razer Basilisk X HyperSpeed as /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.2/0003:1532:0083.0016/input/input43
[  759.283899] razermouse 0003:1532:0083.0016: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Basilisk X HyperSpeed] on usb-0000:00:14.0-8/input2
[  762.103187] audit: type=1101 audit(1625526114.429:145): pid=2731 uid=1000 auid=1000 ses=1 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="brandon" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  762.103314] audit: type=1110 audit(1625526114.429:146): pid=2731 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  762.104739] audit: type=1105 audit(1625526114.429:147): pid=2731 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
```

z3ntu, 2021-07-11:
Then it doesn't look like it's caused by openrazer... You could check with `evtest` or `libinput debug-events` if any events are even coming to user space from the kernel, maybe you find something there.

BrandonHarrisonCode, 2021-11-19:
I wanted to update that this issue appears to have resolved itself on its own. I am closing the issue as it is most likely something to do with something else in the chain.

closed 2021-11-19 by BrandonHarrisonCode