openrazer/openrazer #1531: Manjaro not working with openrazer

AbleTheAbove, 2021-06-18:
### Distribution

Manjaro 

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I tried getting open razer and polychromatic to work and I couldn't

lah7, 2021-06-18:
Need more details... is there an error? What happens? Common missed installation steps are:

* Have you added your user to the `plugdev` group and restarted? (Run `openrazer-daemon -Fv`)

* Have you installed the correct kernel header package? Usually that's `linux-headers`, but if you use the LTS kernel then `linux-lts-headers`. If they were missing, then you'll need to rebuild the driver with `sudo dkms install openrazer-driver/3.0.1`

AbleTheAbove, 2021-06-18:
I've added my user to the group 

I've also installed the headers 
There is a error saying device not valid

lah7, 2021-06-18:
Please could you paste the log / output or a screenshot?

Also, what Razer hardware do you have?

    lsusb | grep 1532

AbleTheAbove, 2021-06-19:
[image]

[image]
Sorry about the delay

lah7, 2021-06-19:
That version of Polychromatic doesn't look like it was compiled correctly - the custom style is incorrect - not sure how that happened. (It has an option to enable native Qt themes in **Edit → Preferences** if desired)

Try clicking **Troubleshoot**, this should identify the problem. I'm guessing DKMS or plugdev permission error, since your device is supported.

AbleTheAbove, 2021-06-21:
[image]

z3ntu, 2021-06-21:
Can you please follow the text and run the first command?

AbleTheAbove, 2021-06-24:
[image]

lah7, 2021-06-24:
The kernel headers are missing - install the package `linux510-headers`, run the `dkms install` command again and then reboot.

Your system might be out of date, the latest 5.10 version is `5.10.42` but the output above suggests you're running `5.10.36`. The driver won't install if the kernel and headers have mismatching versions.

So, to get everything back on track:

    sudo pacman -Syu linux510-headers

AbleTheAbove, 2021-06-26:
Thanks that fixed the issue

closed 2021-06-26 by AbleTheAbove