openrazer/openrazer #1533: Support for Kraken V3 X [1532:0537]

tulanian, 2021-06-21:
### Device Info

* Name: Kraken V3 X 
* 
* Product Number:RZ04-03750100-R3M1
* Model Number: RZ04-375

### Device Information
```
Please post the output of
Bus 003 Device 005: ID 1532:0537 Razer USA, Ltd Razer Kraken V3 X
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0537 
  bcdDevice            0.06
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken V3 X
  iSerial                 3 00000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0123
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0072
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               6 Sidetone
      AudioControl Interface Descriptor:
        bLength                10
[... 255 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0025  1x 37 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```

```
Please post the output of
drwxr-xr-x 2 root root 240 Jun 19 13:43 .
drwxr-xr-x 4 root root 700 Jun 19 13:43 ..
lrwxrwxrwx 1 root root  10 Jun 19 10:46 usb-046d_HD_Pro_Webcam_C920_A8E4B6BF-event-if00 -> ../event17
lrwxrwxrwx 1 root root   9 Jun 19 10:46 usb-Dell_Dell_Universal_Receiver-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 Jun 19 10:46 usb-Dell_Dell_Universal_Receiver-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Jun 19 10:46 usb-Dell_Dell_Universal_Receiver-if01-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Jun 19 10:46 usb-Dell_Dell_Universal_Receiver-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jun 19 10:46 usb-Lenovo_ThinkPad_Thunderbolt_3_Dock_USB_Audio_000000000000-event-if03 -> ../event8
lrwxrwxrwx 1 root root   9 Jun 19 10:46 usb-Logitech_USB_Receiver-if01-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Jun 19 10:46 usb-Logitech_USB_Receiver-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Jun 19 11:10 usb-Razer_Razer_Kraken_V3_X_00000000-event-if03 -> ../event26
lrwxrwxrwx 1 root root  10 Jun 19 10:46 usb-SunplusIT_Inc_Integrated_Camera-event-if00 -> ../event14

```

1337ingDisorder, 2022-08-12:
Any ETA for support for this device?

PabloGirol, 2023-08-11:
I may provide the pcap files needed, but im not really experienced on it, would that be enough to add support for this device?

z3ntu, 2023-08-13:
When providing pcaps I can give some good pointers what parameters are used by the device so you (or someone else) can add support for the device, see https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

V10lator, 2024-06-16:
@z3ntu Tried to get some pcaps but what I get is completely different to what's seen in the linked tutorial, like the protocol looks completely different to me.

For example here's me just toggling between these two states multiple times:
[image]
[image]
[on-off.zip](https://github.com/user-attachments/files/15855675/on-off.zip)

Is the protocol really that different or am I doing something wrong?

z3ntu, 2024-07-02:
[image]

If it's actually this data, then yes it's different... Or the capture didn't capture the correct interface potentially? I'd actually be surprised if they invented a completely new way for this device but I can't rule out anything.

g3t0r, 2025-02-10:
I've found usage of exactly the same protocol with Kitty V2 Headset (1532:0560). Maybe they are making it a new standard? It makes effects rely only on the host software

kingcavespider1, 2025-06-11:
I have one, but it is damaged so not sure how long I will be using it, do you still need packet captures?