openrazer/openrazer #1534: not recognizing my devices

JPcalkins213, 2021-06-21:
So i have an all razer setup with mostly current peripherals. i have a razer lancehead TE razer blackwidow and a razer nari headset and the application is not seeing any of the devices

lah7, 2021-06-22:
Please let us know your distro and application _(Polychromatic, RazerGenie)_ you're using?

Perhaps your user account wasn't added to the `plugdev` group during installation (requires a log out and log in to take effect):

    sudo gpasswd -a $USER plugdev

If that doesn't do the trick, [please follow the troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting) and report back which command is failing - in that case, the DKMS driver is not installed or loading properly.

Once installed correctly, all your devices should work except the Razer Nari (#724) which is not supported yet.

rakhmanovda, 2021-06-23:
I have similar issue. I am using Naga Trinity, it is discoverable as usb device
Bus 001 Device 006: ID 1532:0067 Razer USA, Ltd ITE Device(8595)

every step of the troubleshooting guide seems to be fine
~$ sudo dkms install openrazer-driver/2.5.0
Module openrazer-driver/2.5.0 already installed on kernel 5.4.0-58-generic/x86_64

~$ openrazer-daemon -Fv
Unable to lock on the pidfile.

the device is not visible under razercommander and razerGenie

z3ntu, 2021-06-25:
@rakhmanovda You're using an openrazer version from two years ago, so please install the latest one with the instructions on our website: https://openrazer.github.io/#ubuntu

closed 2021-11-24 by github-actions[bot]