openrazer/openrazer #1537: "Unit openrazer-daemon.service not found" after recent reboot!

tromlet, 2021-06-23:
### Distribution

Fedora 33 x86-64

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

I used to be able to middle-click with my Razer DeathAdder Essential (ID 006E, which is supported), but I was suddenly unable to AFTER a restart! It seems like the OpenRazer service (`openrazer-daemon.service`) has simply gone away on my system, and I am unable to restart it - it doesn't seem like it's present in my `/etc/systemd` folder anywhere. The `openrazer-daemon` COMMAND still lives on my system, but I have no idea what to do with it or how to create a new service daemon - this might be good information to have on that troubleshooting page.

Honestly it worked great on my initial install, but now it doesn't appear to be working, and I'm getting some weird feedback:

**dmesg:**
```log
[   13.568600] razermouse: loading out-of-tree module taints kernel.
[   13.570781] razermouse: module verification failed: signature and/or required key missing - tainting kernel
[   13.591205] razermouse 0003:1532:006E.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer DeathAdder Essential] on usb-0000:00:1a.0-1.4/input0
[   13.677033] razermouse 0003:1532:006E.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer DeathAdder Essential] on usb-0000:00:1a.0-1.4/input1
[   13.789597] razermouse 0003:1532:006E.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Essential] on usb-0000:00:1a.0-1.4/input2
```

`cat ~/.local/share/openrazer/logs/razer.log`:
```log
2021-06-23 13:35:25 | razer                          | CRITICAL | User is not a member of the plugdev group
2021-06-23 13:35:25 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2021-06-23 14:04:36 | razer                          | CRITICAL | User is not a member of the plugdev group
2021-06-23 14:04:36 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2021-06-23 14:12:42 | razer                          | CRITICAL | User is not a member of the plugdev group
2021-06-23 14:12:42 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
```

`sudo journalctl -b | grep -i razer_mount`:
```log
Jun 23 14:53:53 apollo razer_mount[1001]: Driver razermouse
Jun 23 14:53:53 apollo razer_mount[1002]: Device_ID 1-1.4
Jun 23 14:53:53 apollo razer_mount[1003]: Modprobing razermouse
Jun 23 14:53:53 apollo razer_mount[1042]: Modprobed razermouse
Jun 23 14:53:53 apollo razer_mount[1047]: Driver razermouse
Jun 23 14:53:53 apollo razer_mount[1049]: Driver razermouse
Jun 23 14:53:53 apollo razer_mount[1051]: Device_ID 1-1.4:1.1
Jun 23 14:53:53 apollo razer_mount[1052]: Device_ID 1-1.4:1.0
Jun 23 14:53:53 apollo razer_mount[1046]: Driver razermouse
Jun 23 14:53:53 apollo razer_mount[1055]: Device_ID 1-1.4:1.2
Jun 23 14:53:53 apollo razer_mount[1056]: Driver razermouse
Jun 23 14:53:53 apollo razer_mount[1057]: Driver razermouse
Jun 23 14:53:53 apollo razer_mount[1058]: Device_ID 0003:1532:006E.0002
Jun 23 14:53:53 apollo razer_mount[1060]: Device_ID 0003:1532:006E.0001
Jun 23 14:53:53 apollo razer_mount[1061]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:006E.0002/
Jun 23 14:53:53 apollo razer_mount[1062]: Driver razermouse
Jun 23 14:53:53 apollo razer_mount[1063]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:006E.0001/
Jun 23 14:53:53 apollo razer_mount[1066]: Device_ID 0003:1532:006E.0003
Jun 23 14:53:53 apollo razer_mount[1069]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:006E.0003/
Jun 23 14:53:53 apollo razer_mount[1070]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:006E.0002/
Jun 23 14:53:53 apollo razer_mount[1071]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:006E.0001/
Jun 23 14:53:53 apollo razer_mount[1073]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:006E.0003/
```

THIS is interesting output on the kernel side, maybe this is conflicting?
```shell
$ uname -r
5.12.11-200.fc33.x86_64
$ sudo dkms status
[sudo] password for tom: 
openrazer-driver, 3.0.1, 5.11.10-200.fc33.x86_64, x86_64: installed
openrazer-driver, 3.0.1, 5.12.11-200.fc33.x86_64, x86_64: installed
```

And yet, when I run my `groups` command, I get the following output: `tom wheel dialout plugdev`, and I HAVE rebooted, but I'll try again. :P

lah7, 2021-06-24:
> razermouse: module verification failed: signature and/or required key missing - tainting kernel

This would suggest the driver is not loading at boot -- in most cases it's because secure boot is enabled in the EFI settings -- could a recent update have turned it back on? I'm not familiar with Fedora myself.

I think the `plugdev` compliant could be as a result of the driver being missing.

z3ntu, 2021-06-25:
Also we're shipping a systemd **user** unit with the package, that's why you need to use `--user` with the systemctl commands, e.g. `systemctl --user status openrazer-daemon.service`

closed 2021-11-24 by github-actions[bot]