openrazer/openrazer #1538: Razer 14 2021 - issues

koen81, 2021-06-27:
Based on my own experience and on the experience of others from this thread:

https://www.reddit.com/r/linuxhardware/comments/o2ec0d/amd_razer_blade_14_2021_first_few_hours_on_linux/

Issues I noticed;

1. Brightness stuck at 100%
2. Fractural scaling not working for other settings then 100% en 200%

Issues mentioned in thread;

1. Headphone jack works edit: stopped working for me. I most likely didn't test it thoroughly enough ootb, since I don't really use the headphone jack
2. video out only works in NVIDIA graphics mode
There is no video out when running Hybrid or integrated graphics
This was tested on the built-in HDMI port, as well as with a usb-c hub on all the usb-c ports

Device information (RTX 3070 with QHD screen 165Hz);

H/W path          Device        Class       Description
=======================================================
                                system      Computer
/0                              bus         Motherboard
/0/0                            memory      15GiB System memory
/0/1                            processor   AMD Ryzen 9 5900HX with Radeon Graph
/0/100                          bridge      Renoir Root Complex
/0/100/0.2                      generic     Renoir IOMMU
/0/100/1.1                      bridge      Renoir PCIe GPP Bridge
/0/100/1.1/0                    display     NVIDIA Corporation
/0/100/1.1/0.1                  multimedia  NVIDIA Corporation
/0/100/2.2                      bridge      Renoir PCIe GPP Bridge
/0/100/2.2/0      wlp2s0        network     Intel Corporation
/0/100/2.4                      bridge      Renoir PCIe GPP Bridge
/0/100/2.4/0                    storage     NVMe SSD Controller SM981/PM981/PM98
/0/100/2.4/0/0    /dev/nvme0    storage     SAMSUNG MZVLB1T0HBLR-00A00
/0/100/2.4/0/0/1  /dev/nvme0n1  disk        NVMe namespace
/0/100/8.1                      bridge      Renoir Internal PCIe GPP Bridge to B
/0/100/8.1/0                    display     Advanced Micro Devices, Inc. [AMD/AT
/0/100/8.1/0.2                  generic     Family 17h (Models 10h-1fh) Platform
/0/100/8.1/0.3                  bus         Renoir USB 3.1
/0/100/8.1/0.4                  bus         Renoir USB 3.1
/0/100/8.1/0.5                  multimedia  Raven/Raven2/FireFlight/Renoir Audio
/0/100/8.1/0.6                  multimedia  Family 17h (Models 10h-1fh) HD Audio
/0/100/14                       bus         FCH SMBus Controller
/0/100/14.3                     bridge      FCH LPC Bridge
/0/101                          bridge      Renoir PCIe Dummy Host Bridge
/0/102                          bridge      Renoir PCIe Dummy Host Bridge
/0/103                          bridge      Renoir PCIe Dummy Host Bridge
/0/104                          bridge      Advanced Micro Devices, Inc. [AMD]
/0/105                          bridge      Advanced Micro Devices, Inc. [AMD]
/0/106                          bridge      Advanced Micro Devices, Inc. [AMD]
/0/107                          bridge      Advanced Micro Devices, Inc. [AMD]
/0/108                          bridge      Advanced Micro Devices, Inc. [AMD]
/0/109                          bridge      Advanced Micro Devices, Inc. [AMD]
/0/10a                          bridge      Advanced Micro Devices, Inc. [AMD]
/0/10b                          bridge      Advanced Micro Devices, Inc. [AMD]
/0/2                            system      PnP device PNP0c01
/0/3                            system      PnP device PNP0b00
/0/4                            system      PnP device PNP0c02

z3ntu, 2021-06-27:
Interesting but these problems don't seem to be related to openrazer at all... Not sure what you're expecting me to do?

shizonic, 2021-06-27:
Isn't it necessary to add this model?

z3ntu, 2021-06-27:
It won't solve any of the issues mentioned in the first comment but it will be possible to control the rgb lighting on the device. Please provide the `lsusb | grep 1532` output.

shizonic, 2021-06-27:
> It won't solve any of the issues mentioned in the first comment but it will be possible to control the rgb lighting on the device. Please provide the `lsusb | grep 1532` output.

I see. Thank you.

I am still waiting for my device so @koen81 may you provide the requested infos?

chriswatts91, 2021-06-27:
I have a Razor 14 2021, heres `lsusb | grep 1532` output:

`Bus 001 Device 003: ID 1532:0270 Razer USA, Ltd Razer Blade`

z3ntu, 2021-06-28:
@chriswatts91 Can you please [open a device support issue](https://github.com/openrazer/openrazer/issues/new?assignees=&labels=Device+Support&template=new-device-support.md&title=Support+for+%5Bdevice+name+here%5D) for this device?

chriswatts91, 2021-06-28:
@z3ntu Sure: #1540

z3ntu, 2021-06-30:
Thanks. I'll be closing this issue because it's not related to openrazer but feel free to continue chatting here (if you don't find a better place for the conversation)

closed 2021-06-30 by z3ntu