openrazer/openrazer #1540: Support for Razor Blade 14

chriswatts91, 2021-06-28:
### Device Info

* Name: Razor Blade 14 (Early 2021)
* Product Number: RZ09-0370AW23
* Model Number: RZ09-0370

### Device Information
```
Bus 001 Device 003: ID 1532:0270 Razer USA, Ltd Razer Blade
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0270 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
total 0
drwxr-xr-x 2 root root 180 Jun 28 12:22 .
drwxr-xr-x 4 root root 600 Jun 28 12:22 ..
lrwxrwxrwx 1 root root  10 Jun 28 00:03 usb-Azurewave_Integrated_Camera_200901010001-event-if00 -> ../event13
lrwxrwxrwx 1 root root  10 Jun 28 00:03 usb-Azurewave_Integrated_Camera_200901010001-event-if02 -> ../event14
lrwxrwxrwx 1 root root  10 Jun 28 00:03 usb-Razer_Razer_Blade-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 Jun 28 00:03 usb-Razer_Razer_Blade-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Jun 28 00:03 usb-Razer_Razer_Blade-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Jun 28 00:03 usb-Razer_Razer_Blade-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 Jun 28 00:03 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
```

### Packet Captures

### Synapse Screenshots

BentonEdmondson, 2021-07-06:
Is there any more info needed to get this going?

chriswatts91, 2021-07-07:
> Is there any more info needed to get this going?

Packet Captures I think. I'm struggling to find time to add them; and it's not something I've done before.

https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

jbdutton, 2021-08-21:
I was able to get the Wireshark / Synapse set-up working on my Blade 14. Is there a particular routine (e.g. set this color, change this row) that I could capture to help out? Attached are a few random changes, like toggling lighting off and on, setting specific keys to a color, etc. Please let me know how I can help.

[openrazer-packets.zip](https://github.com/openrazer/openrazer/files/7024528/openrazer-packets.zip)

mrmetaverse, 2021-08-29:
Could we merge this for everyone?

PR #1586 (2021-10-03): Add support for Razer Blade 14 (2021)
Works as expected, no issues found.
fixes #1540
closed 2021-10-03 by z3ntu