openrazer/openrazer #1542: Support for Razer Blade 15 Advanced (Mid 2021)

matmarchant, 2021-06-29:
### Device Info

* Name: Razer Blade 15 Advanced - Mid 2021 
* Product Number: RZ09-0409BWD3
* Model Number: RZ09-0409

### Device Information
```
lsusb -d 1532: -v

Bus 003 Device 003: ID 1532:0276 Razer USA, Ltd Razer Blade
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0276 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

```
lrwxrwxrwx 1 root root   9 Jun 29 10:54 usb-Azurewave_Integrated_RGB_Camera_01.00.00-event-if00 -> ../event4
lrwxrwxrwx 1 root root   9 Jun 29 10:54 usb-Azurewave_Integrated_RGB_Camera_01.00.00-event-if02 -> ../event5
lrwxrwxrwx 1 root root  10 Jun 29 10:54 usb-Razer_Razer_Blade-event-if01 -> ../event10
lrwxrwxrwx 1 root root   9 Jun 29 10:54 usb-Razer_Razer_Blade-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Jun 29 10:54 usb-Razer_Razer_Blade-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root  10 Jun 29 10:54 usb-Razer_Razer_Blade-if02-event-mouse -> ../event11
lrwxrwxrwx 1 root root   9 Jun 29 10:54 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

matmarchant, 2021-07-09:
Closing as due to multiple issues I had to return the laptop

z3ntu, 2021-07-09:
Please keep the issue open as long as the device is not supported. If you wish, you can unsubscribe from the issue on the right side so you won't get any more notifications.

j-maynard, 2021-07-17:
@z3ntu I've raised a PR for this which seems to work on my new Blade.  You can take a look at:

https://github.com/openrazer/openrazer/pull/1561

j-maynard, 2021-08-07:
So we have a working PR for basic lighting but I've since discoverfed Razer Synapse on this Blade also supports a "Breathing Logo".  I've raise support for that as a seperate issue so we can get basic support for this device quickly and revisit the breathing logo when we have some packet captures.  The issue for the brething logo can be found at: https://github.com/openrazer/openrazer/pull/1561

PR #1561 (2021-08-15): Add support for Razer Blade Advanced 15 Mid 2021
This is an initial commit to support the Razer Blade Advanced 15 Mid 2021 release.  This PR was tested on the 4k model of the Blade Advanced and I can confirm changing keyboard color and most profiles other than Startlight (random) seem to work.  I can't find the references to the logo to allow logo control to work.  If someone can point me in the right direction I might be able to fix that.

Fixes #1542
closed 2021-08-15 by z3ntu