openrazer/openrazer #1543: Is there a way to disable the crossfade effect transition?

hacker1024, 2021-07-01:
### Distribution

macOS Big Sur 11.4 (20F71)

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm using https://github.com/1kc/openrazer, which consists of OpenRazer's kernel driver code adapted to macOS. I think my question is applicable here.

When I apply a keyboard effect using a function like `razer_attr_write_mode_static` in `razerkbd_driver.c`, a crossfade is used to animate the change. Is there a way to stop this, and switch to the new effect immediately?

I'm implementing Razer's Chroma REST API server, so I need behaviour that matches Windows.

lah7, 2021-07-01:
Since static is a hardware effect, there is no further control over what the firmware is doing (fading static to another static).

A trick is to switch to another effect (such as wave, or draw a blank custom frame) and then immediately switch back to static, which immediately changes colour without the fade.

hacker1024, 2021-07-01:
Thanks for the tip. The Windows SDK can definitely do it without a crossfade, though, and there's no noticeable effect inbetween...

closed 2021-07-01 by hacker1024