openrazer/openrazer #1544: Support for [device name here]

Dingskeheezeg, 2021-07-01:
### Device Info

* Name: <BlackWidow V3 mini hyperspeed>
* Product Number: <RZ03-03890300>
* Model Number: <RZ03-0389>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
in macos i have no lusb, but device id is 0x0271 via wireless dongle, 0x258 connected with the usb wire
```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
this is the one : 

https://www.razer.com/gaming/keyboards/razer-blackwidow-v3-mini-hyperspeed

MagnusStrom, 2021-07-02:
I have the same keyboard, the output of `lsusb -d 1522: -v` is:
```            Razer BlackWidow V3 Mini:

              Product ID: 0x0258
              Vendor ID: 0x1532
              Version: 2.00
              Speed: Up to 12 Mb/s
              Manufacturer: Razer
              Location ID: 0x14140000 / 4
              Current Available (mA): 500
              Current Required (mA): 500
              Extra Operating Current (mA): 0```

Dingskeheezeg, 2021-07-02:
i guess its almost the same info i found in the systeminfo of my mac

Razer BlackWidow V3 Mini:

  Productcode:	0x0271
  Fabrikantcode:	0x1532
  Versie:	1.00
  Serienummer:	000000000000
  Snelheid:	Maximaal 12 Mb/s
  Fabrikant:	Razer
  Locatiecode:	0x14222200 / 9
  Beschikbare stroom (mA):	500
  Vereiste stroom (mA):	100
  Extra stroom in werking (mA):	0

The 0x258 device id is when connected with the usb wire directly, mine is connected withe wireless dongle supplied with it, normally there should be an  onther device id for the connection over BT

Dingskeheezeg, 2021-07-02:
the device id connected over bluetooth is 0x272, the keyboard has three different profiles to chose when connecting over bluetooth, all three are 0x272

z3ntu, 2021-07-03:
To summarize the USB IDs:
* `1532:0258` USB
* `1532:0271` Wireless
* `1532:0272` Bluetooth

Dingskeheezeg, 2021-07-03:
yes

thx

Dingskeheezeg, 2021-07-03:
Would it be possible to also add some more advanced lithning effects that synapse 3 provide ?
Like maybe the starlight or fire , or maybe an customized wave effect ?

thx in advance ( it would make me consider to be happy to sponsor some more $ )

z3ntu, 2021-07-05:
OpenRazer provides the infrastructure for you to write the effects yourself, we have an example for a [custom starlight effect in the repo](https://github.com/openrazer/openrazer/blob/master/examples/custom_starlight.py). We do not have integrated support for custom effects yet though.

Dingskeheezeg, 2021-07-08:
what is the proigress on this ?

thx

Dingskeheezeg, 2021-07-26:
how about progfress on this ?

MagnusStrom, 2021-07-26:
progress <3

tammysalmon, 2021-07-29:
I'm wondering if [this task](https://github.com/openrazer/openrazer/commit/a6d227ebced01fca4bb2dfa09acf419850161f93) gets us pretty close to support here. We just need the updated USB PID.

Dingskeheezeg, 2021-07-30:
when will this be added ? i have no problem , me having donated nice amount $ for this , but now nothing happening ...

i guess there are more people then me who use the blackwidow v3 keyboards

z3ntu, 2021-08-01:
Please create a packet capture of Razer Synapse interacting with the device: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

MagnusStrom, 2021-08-01:
Can I do this on bootcamp?

Dingskeheezeg, 2021-08-06:
> Please create a packet capture of Razer Synapse interacting with the device: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

the thing is , i don't have an windows at hand, therefore i donated a nice amount of $ for this, so ...

MagnusStrom, 2021-08-11:
I have windows. Going to run this.

MagnusStrom, 2021-08-11:
Nvm, I might not be able to. I'm willing to paypal someone money to get this working tho

Dingskeheezeg, 2021-08-11:
> Nvm, I might not be able to. I'm willing to paypal someone money to get this working tho

then they take your money and do nothing apparently ... is donated around 50$ , like more then a month a go ... so still nothing

Dingskeheezeg, 2021-09-11:
so when will the blackwidow v3 mini device finally be added ?

IvanVojtko, 2021-09-24:
Hey guys. I just bought this keyboard and added support for it in this PR: https://github.com/openrazer/openrazer/pull/1615.
 If you build this PR, you should be able to control backlight but only while connected using USB cable.It remembers your settings once you disconnect the cable so should be fine for now. I will try to add support for wireless control but in the meantime I think it's better than nothing. Let me know if you face any issues.

MagnusStrom, 2021-09-24:
> Hey guys. I just bought this keyboard and added support for it in this PR: #1615.
> If you build this PR, you should be able to control backlight but only while connected using USB cable.It remembers [...]

What a chad. Right as my macos corrupts itself and im forced to windows lmfaoo
good job tho!

IvanVojtko, 2021-09-25:
Just an update. I spent few more hours on this and added support for wireless connection as well and also created a correct mapping of Fn keys. Everything should be working fine now, you will even get a battery notifications. https://github.com/openrazer/openrazer/pull/1617

closed 2021-10-03 by z3ntu

ashenjoseph, 2022-01-17:
Hi, I am having trouble connecting this with RazerMacOS, the keyboard can't seem to be detected. I have tried with wired and 2.4 wireless. Is there something I need to do before the settings can be accessed?

z3ntu, 2022-01-17:
This project is not targeting macOS so we can't help you with that.