openrazer/openrazer #1545: Razer Kiyo Pro?

AwesomeKenP, 2021-07-02:
Hi,

Is there a roadmap to support Razer webcams on Linux? So far, the apps I have don't allow much control of all its awesome features.

Thanks!

z3ntu, 2021-07-03:
Please see https://github.com/openrazer/openrazer/issues/1328#issuecomment-798452048

closed 2021-07-03 by z3ntu

cdgriffith, 2021-10-30:
I think this is worth re-opening as the Kiyo Pro (different from regular Kiyo in the linked PR) has specific settings that can only be changed through Synapse. Those are HDR and Field of view. 

I haven't ever worked with usb capture and sending raw settings before but took an initial stab at it by using wireshark to see the request sent when enabling HDR. 

[image]

I tried to implement what it was doing with python, but it seems the setting does not save, I'm doing it wrong, or there is something else going on as it is still not in HDR mode. 

```python
# Requires PyUSB installed
import usb.core
import usb.util

dev = usb.core.find(idVendor=0x1532, idProduct=0x0e05)

if dev is None:
    raise ValueError('Device not found')

print(dev.get_active_configuration())

if dev.is_kernel_driver_active(0):
    dev.detach_kernel_driver(0)

if dev.is_kernel_driver_active(1):
    dev.detach_kernel_driver(1)

usb.util.claim_interface(dev, 0)
usb.util.claim_interface(dev, 1)

dev.set_interface_altsetting(interface=1, alternate_setting=0)

usb.util.release_interface(dev, 0)
usb.util.release_interface(dev, 1)

dev.attach_kernel_driver(0)
# trying to reattach the second kernel driver raises error
```

Viewing raw stream with ffplay: 

```
ffplay -f v4l2 -input_format nv12 -video_size 1920x1080 -framerate 30 -i /dev/video0
```

z3ntu, 2021-10-31:
If the device doesn't use the same or a similar protocol as other Razer devices (which I don't imagine it does) then I  don't consider this device to be in scope of openrazer. If you upload the USB packet captures I can quickly take a look at them and confirm.

cdgriffith, 2021-11-22:
Thanks for offering to look @z3ntu! Hoping I did this right, was in a windows VM on linux host with USB passthrough to try and reduce noise. 

[kiyo_pro_pcaps.zip](https://github.com/openrazer/openrazer/files/7582102/1620-2A60.zip)