openrazer/openrazer #1550: daily builds / install from master - debian based system

a-simoes, 2021-07-07:
Hey.
I noticed that master branch contains support for the Razer Basilisk Essential.
And the latest build for debian (3.0.1) was made before this support was introduced.

I want to know if there is any daily build for debian available or any tutorial about how to manually build the binaries from master.

Thanks in advance!

z3ntu, 2021-07-07:
Hi, you can follow https://github.com/openrazer/openrazer/wiki/Building-a-package#debian-9 . We do not have automated daily packages for Debian (we do just for Ubuntu, and on Arch Linux you can use the openrazer-git package). Please let me know if that helps.

a-simoes, 2021-07-07:
Thanks @z3ntu. 

The instructions failed at first time due to unmet dependencies.
I installed dh-python and repeated the process.

It is working now as expected.

Again, thanks a lot!

closed 2021-07-07 by a-simoes