openrazer/openrazer #1552: Fn + Media Keys don't work, some effects don't work either - Razer Ornata Chroma V2

mcpeixoto, 2021-07-08:
Hello there, new to OpenRazer

Just installed OpenRazer, I'm using Snake as a GUI

The thing is, my media keys and my fn + (Macro, Gaming Mode, Brightness) don't work.. :/
Plus, there are some effect's that also don't work such as: Reactive, Ripple and Audio (The keyboard is just black)

I'm using Kubuntu (Ubuntu 20.04.2 LTS)
openrazer-daemon 3.0.1

gardockt, 2021-07-08:
I have a similar issue on Arch Linux with Polychromatic. In my case _Reactive_ mode started working consistently after a few retries, however _Ripple_ simply disables the backlight. While the daemon is running, Macro/Gaming Mode/Brightness/turning volume knob are recognized by `xev` as unknown buttons, example for "brightness down" key:

> FocusOut event, serial 35, synthetic NO, window 0xe00001,
>     mode NotifyGrab, detail NotifyAncestor
> 
> FocusIn event, serial 35, synthetic NO, window 0xe00001,
>     mode NotifyUngrab, detail NotifyAncestor
> 
> KeymapNotify event, serial 35, synthetic NO, window 0x0,
>     keys:  4294967227 0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
>            0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0  

And for "brightness up" key:

> KeyPress event, serial 35, synthetic NO, window 0xe00001,
>     root 0x6bb, subw 0x0, time 9205084, (196,-282), root:(1167,277),
>     state 0x0, keycode 202 (keysym 0x0, NoSymbol), same_screen YES,
>     XLookupString gives 0 bytes: 
>     XmbLookupString gives 0 bytes: 
>     XFilterEvent returns: False
> 
> KeyRelease event, serial 35, synthetic NO, window 0xe00001,
>     root 0x6bb, subw 0x0, time 9205143, (196,-282), root:(1167,277),
>     state 0x0, keycode 202 (keysym 0x0, NoSymbol), same_screen YES,
>     XLookupString gives 0 bytes: 
>     XFilterEvent returns: False

Multimedia keys and pressing volume knob type characters depending on how many of them are pressed simultaneously:
1. _bcd_
2. _hjkl_
3. _mnprst_
4. _uvwxz12_

Killing the daemon makes these buttons work as intended.

mcpeixoto, 2021-07-09:
Thanks for the information @gardockt, now at least I have a janky workaround! 
1) Select the light mode you want (e.g. wave)
2) Kill the daemon -> The media keys work as intended and the light mode chosen persists!

Some notes: When the daemon is running my FN+Brightness Down key mutes/unmutes the microphone and my scroll volume wheel works as a mouse wheel

autotechmusic, 2021-07-21:
Greetings all,

Same here.  My biggest issue is the volume wheel.  It is indeed a mouse wheel.  The click on the wheel is "bcd".
Killing the daemon works in the meantime.  The problem with this keyboard is that it has no memory.  The old Black Widow remembers my settings even when unplugged for days.  This one doesn't hold anything at all.  It appears as though they nerfed the Ornata v2, and perhaps others, to "need" synapse for full function.  Doesn't seem right.
Thanks for the daemon kill tip.

AT

z3ntu, 2021-07-25:
Volume wheel will be fixed by #1547

SamantazFox, 2021-11-04:
> Multimedia keys and pressing volume knob type characters depending on how many of them are pressed simultaneously:
> 
>     1. _bcd_
>     2. _hjkl_
>     3. _mnprst_
>     4. _uvwxz12_
> 
> Killing the daemon makes these buttons work as intended.

The exact same problem happens with my Ornata v2 (VID/PID `1532:025D`).

One odd thing is that it worked for half a day, and suddently it wasn't working at all. And killing the daemon restores full functionality of the 3 media keys ("previous", "play/pause" and "next").

michihupf, 2021-12-16:
Can confirm media keys (⏮, ⏯, ⏭) do not work, volume wheel does though. 
Platform: Garuda Linux (Arch)
Keyboard: Razer Ornata Chroma V2

closed 2022-07-20 by github-actions[bot]