openrazer/openrazer #1553: Support for Razer Orochi v2

ccxex29, 2021-07-08:
### Device Info

* Name: Orochi v2  
*Razer Synapse shows this as Razer Orochi v2 White Edition*
* Product Number: RZ01-03730400
* Model Number: RZ01-0373

### Device Information
Please post the output of
    `lsusb -d 1532: -v`  

```
Bus 001 Device 012: ID 1532:0094 Razer USA, Ltd Orochi V2
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0094 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Orochi V2
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        4
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     325
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1  
```

Please post the output of
    `ls -lah /dev/input/by-id/`  

```
lrwxrwxrwx    10 root  8 Jul 17:51  usb-Razer_Orochi_V2_000000000000-event-if01 -> ../event24
lrwxrwxrwx    10 root  8 Jul 17:51  usb-Razer_Orochi_V2_000000000000-event-if03 -> ../event26
lrwxrwxrwx    10 root  8 Jul 17:51  usb-Razer_Orochi_V2_000000000000-event-mouse -> ../event21
lrwxrwxrwx    10 root  8 Jul 17:51  usb-Razer_Orochi_V2_000000000000-if01-event-kbd -> ../event22
lrwxrwxrwx    10 root  8 Jul 17:51  usb-Razer_Orochi_V2_000000000000-if01-event-mouse -> ../event23
lrwxrwxrwx     9 root  8 Jul 17:51  usb-Razer_Orochi_V2_000000000000-if01-mouse -> ../mouse5
lrwxrwxrwx    10 root  8 Jul 17:51  usb-Razer_Orochi_V2_000000000000-if02-event-kbd -> ../event25
lrwxrwxrwx     9 root  8 Jul 17:51  usb-Razer_Orochi_V2_000000000000-mouse -> ../mouse4
```

### Packet Captures
I am unsure about how I should do it on a non-RGB mouse, but [here](https://drive.google.com/drive/folders/1Wyp4oh7uGKeWJxqleWIrSJSlTdyD7NYX?usp=sharing) is a gdrive folder with pcapng for each click. Let me know if anything is missing.  
https://drive.google.com/drive/folders/1Wyp4oh7uGKeWJxqleWIrSJSlTdyD7NYX?usp=sharing

### Synapse Screenshots
[image]
[image]
[image]
[image]

z3ntu, 2021-07-11:
@ccxex29 As this device is fairly trivial, I've implemented support for it in #1555, please test!

Also according to the product page you can connect this device via both the USB dongle and via Bluetooth. I'm guessing you had it connected via the dongle, so please also provide the lsusb output when connected via Bluetooth.

Regarding the "White edition" variant, there is *some* indication in the pcaps what logic Synapse is using for that but so far we don't have any support for device variants (basically just a name change) in openrazer so it'll just show up as Orochi V2 which should be good enough.

ccxex29, 2021-07-12:
I'm not really familiar with razer peripherals but, the average polling rate is always 500Hz with some spikes to 1000Hz on all options when I monitored it using evhz (probably the behaviour without proper driver?).

I used razercommander gui to test this mouse, but it seems that the device can only be detected but applying both the polling rate and dpi, nothing changes.  

[image]

About the bluetooth, inside `/sys/devices/virtual/misc/uhid/` the mouse is shown as `0005:1532:0095.0018`. Let me know what other informations do I need to gather.

notsukamto, 2021-07-30:
Any update on this?

z3ntu, 2021-08-01:
@notsukamto Please test #1555

luiszimmermann, 2021-10-05:
> @notsukamto Please test #1555

Hello @z3ntu , I tried your branch and the mouse is now detected but the changes in openrazer (using razergenie) dont get applied.

whitewillow, 2021-10-05:
Same for me - @z3ntu - I also testet using Polychromatic - and changes where not saved to the mouse.

z3ntu, 2021-10-06:
Does the serial number get displayed properly? Try `cat /sys/bus/hid/drivers/razermouse/0*/device_serial`

whitewillow, 2021-10-07:
> at /sys/bus/hid/drivers/razermouse/0*/device_serial

Same for me - also empty

luiszimmermann, 2021-11-01:
> @luiszimmermann Please rebuild with the latest commit [887a2c1](https://github.com/openrazer/openrazer/commit/887a2c1e40e1e3704966cbb68950b9add7c987d6) (and reboot) and check the device_serial again. I haven't touched the rest, just the serial one.

@z3ntu Now the `device_serial` is with the right string (the same that it's in the mouse), but the dmesg has the exact same output. And the `razergenie` still doesn't works as expected.
[image]

z3ntu, 2021-11-14:
@luiszimmermann I've added the change to the other methods, so most things should work now, but I'm unsure if `charge_low_threshold` and `device_idle_time` work. Please build from latest commit in #1555 again (49a4187)

luiszimmermann, 2021-11-23:
@z3ntu Tested again with `razergenie` and your commit (https://github.com/openrazer/openrazer/commit/49a4187941c4903752d4407e2e5ec807beb5db08), and now the Polling rate setting works! But the DPI does nothing. I'm using the 2.4 mode with the dongle, didn't tried in BT mode.
This mouse has 4 profiles in memory with the DPI settings, don't know if this affects `openrazer`, I don't have any other Razer mouse, this is my first.
`dmesg` still outputs the same `Command timed out`.

z3ntu, 2021-12-04:
@luiszimmermann Let's please move the testing discussion to the PR #1555

PR #1555 (2022-01-15): Add support for Razer Orochi V2
Fixes #1553

---

Building from this PR: https://github.com/openrazer/openrazer/wiki/Building-a-package
closed 2022-01-15 by z3ntu