openrazer/openrazer #1556: Razer Ornata Chroma V2 | Multimedia Keys

GivouDev, 2021-07-11:
### Distribution

Keyboard: Rater Ornata Chroma V2
Openrazer Version: 3.0.1
Controller: Polychromatic

### Support

- [ X ] Have you tried rebooting?
- [ X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello Guys,

i have some toubles to get the multimedia key of the razer ornata chroma v2 working.
When i stop the deamon, everything works fine. I can use my volume wheel and my multimedia keys..but when i start the deamon the multimedia keys(backward, pause/play, forward) write the following (bcd,bcd,bcd) and my volume wheel becomes a mouse wheel.

With FN + F2 or FN + F3 i can change volume up / down.....but there shouldnt be anything.
Is there anyway to stop the deamon from controlling my makros?

Thank you :D

sixpounder, 2021-07-13:
I can confirm this. Stopping the daemon solves the issue, starting it up again brings the issue back.

I investigated a little (got the keyboard today) and seems like the play button, for instance, gets mapped to a combination of three keys (B, C and D to be precise). The volume wheel just falls back to scroll events.

z3ntu, 2021-07-14:
Duplicate of #1398

closed 2021-07-14 by z3ntu