openrazer/openrazer #1557: Wrong Input events emited (`REL_WHEEL` instead of `REL_HWHEEL`)

G-M0N3Y-2503, 2021-07-14:
### Distribution
```
Distributor ID:	Ubuntu
Description:	Ubuntu 21.04
Release:	21.04
Codename:	hirsute
```

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I was doing some work with `libevdev` and noticed that I was unable to create horizontal wheel events with my device and instead it would create vertical wheel events.

below is one leftwards scroll click (3 events) and 3 downwards scroll detents (1 event each).
```
sudo evtest /dev/input/by-id/usb-Razer_Razer_Naga_2014-event-mouse 
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x1532 product 0x40 version 0x111
Input device name: "Razer Razer Naga 2014"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 8 (REL_WHEEL)
    Event code 11 (REL_WHEEL_HI_RES)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1626260186.084732, type 2 (EV_REL), code 8 (REL_WHEEL), value -1
Event: time 1626260186.084732, type 2 (EV_REL), code 11 (REL_WHEEL_HI_RES), value -120
Event: time 1626260186.084732, -------------- SYN_REPORT ------------
Event: time 1626260186.135062, type 2 (EV_REL), code 8 (REL_WHEEL), value -1
Event: time 1626260186.135062, type 2 (EV_REL), code 11 (REL_WHEEL_HI_RES), value -120
Event: time 1626260186.135062, -------------- SYN_REPORT ------------
Event: time 1626260186.184612, type 2 (EV_REL), code 8 (REL_WHEEL), value -1
Event: time 1626260186.184612, type 2 (EV_REL), code 11 (REL_WHEEL_HI_RES), value -120
Event: time 1626260186.184612, -------------- SYN_REPORT ------------
Event: time 1626260187.035039, type 2 (EV_REL), code 8 (REL_WHEEL), value -1
Event: time 1626260187.035039, type 2 (EV_REL), code 11 (REL_WHEEL_HI_RES), value -120
Event: time 1626260187.035039, -------------- SYN_REPORT ------------
Event: time 1626260187.825604, type 2 (EV_REL), code 8 (REL_WHEEL), value -1
Event: time 1626260187.825604, type 2 (EV_REL), code 11 (REL_WHEEL_HI_RES), value -120
Event: time 1626260187.825604, -------------- SYN_REPORT ------------
Event: time 1626260188.061599, type 2 (EV_REL), code 8 (REL_WHEEL), value -1
Event: time 1626260188.061599, type 2 (EV_REL), code 11 (REL_WHEEL_HI_RES), value -120
Event: time 1626260188.061599, -------------- SYN_REPORT ------------
```
but as you can see their all just downward scroll events.

As an attempt to get further understanding, I tried the mouse in "Driver Mode" instead of "Normal Mode", but [as expected](https://github.com/openrazer/openrazer/issues/201) it just disabled all events from the horizontal wheel movement.

Any help or guidance on where I might look for creating a patch for this would be appreciated.

z3ntu, 2021-07-14:
Hi, I'm not sure what you want to achieve? Pressing the mouse wheel left or right is meant for fast-scrolling up and down. I'm not aware of any other functionality that the mouse provides.

lah7, 2021-07-14:
I think it's both a bug and a feature. The mice (by default?) continuously scrolls left/right, but it's also configurable under Razer Synapse (in software?) to be used as a left/right scroll instead. Not sure which one is the default (various threads on the Internet suggests this might had changed at one point)

Presumably the daemon could catch the event and emit the horizontal scrolling? If it is patchable, it would be better to provide an option in `razer.conf` since I do prefer continuous scrolling, but I understand others would want to enable horizontal scrolling too.

G-M0N3Y-2503, 2021-07-15:
> I'm not sure what you want to achieve? 

what I was hoping to achieve was that unique events are generated from pressing the wheel left or right.
After that, I can just remap it to whatever using something like https://github.com/sezanzeb/key-mapper/

> Pressing the mouse wheel left or right is meant for fast-scrolling up and down. I'm not aware of any other functionality that the mouse provides.

Hmmm, I'm new to using Linux with a GUI, is that functionality a Linux convention for mice wheels?
Coming primarily from a Windows GUI background with a variety of mice, it's always been for scrolling horizontally from my experience.

> I think it's both a bug and a feature. The mice (by default?) continuously scrolls left/right, but it's also configurable under Razer Synapse (in software?) to be used as a left/right scroll [...]

If you're interested, I had a look at synapse on Windows and found the following (and unexpected I might add) defaults.  

[image]
[image]
[image]

> Presumably the daemon could catch the event and emit the horizontal scrolling? If it is patchable, it would be better to provide an option in `razer.conf` since I do prefer continuous scrolling, but I understand others would want to enable horizontal scrolling too.

I haven't yet looked into the specifics of this driver just yet, but I assume that it's translating the USB protocol into input events and writing them to the event device. So it would need to be changed there what event is emitted.
as for what it is changed to I don't mind as long as it's different from the other button events on the mice.  
If `REL_HWHEEL` has objections maybe it could be `BTN_BACK` and `BTN_FORWARD` for left and right respectively, though I wouldn't want some other device to now have two of those buttons.

It doesn't sound like there are any known limitations in doing this so I'll look at trying to patch this with a configuration option.

Thanks, everyone.

z3ntu, 2021-07-15:
Ok so if you want to do this, there is some code for other devices that handles these buttons in driver mode (you will likely never get special events in device mode as the firmware just generates sane events there that don't need any special handling by the OS, meaning you will only get normal looking scroll events)
https://github.com/openrazer/openrazer/blob/55036d2d7a88166cd8220bf39ec6d65f96ad2af6/driver/razermouse_driver.c#L3880-L3900
You can then probably make it emit key events (just choose some keys that are not used yet for other things) and then you should get these key events in user space.

commit d5d27667 (2021-10-03): Add REL_HWHEEL support for the Razer Naga 2014

Fixes #1557
closed 2021-10-03 by z3ntu