openrazer/openrazer #1560: Daemon stops working on pressing macro button

adityabisoi, 2021-07-17:
### Distribution
Ubuntu 20.04

### Device information
Bus 001 Device 008: ID 1532:0209 Razer USA, Ltd BlackWidow Tournament Edition Chroma

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

After installation of OpenRazer, my fn+f11 (keyboard brightness down) button has been working like a mic on/off button. 

Also, fn+ Macro and fn+f12 (keyboard brightness up) button on pressing now make the daemon to stop working. I cannot change the keyboard color anymore until I restart the daemon/plug-out and plugin the keyboard again.

I am unable to figure out the problem, as I have tried everything mentioned in the troubleshooting page already. Any help is much appreciated.

z3ntu, 2021-07-25:
Can you check `~/.local/share/openrazer/logs/razer.log` if there's any error?

adityabisoi, 2021-07-25:
Yes, a part of the log which has error is as below
```
2021-07-17 11:29:11 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 23604
2021-07-17 11:29:11 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-07-17 11:29:11 | razer                          | INFO     | Serving DBus
2021-07-17 11:29:24 | razer                          | INFO     | Stopping daemon.
2021-07-17 11:29:25 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 23667
2021-07-17 11:29:25 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-07-17 11:29:25 | razer                          | INFO     | Serving DBus
2021-07-17 11:29:47 | razer                          | INFO     | Stopping daemon.
2021-07-17 11:29:49 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 23904
2021-07-17 11:29:49 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-07-17 11:29:49 | razer                          | INFO     | Serving DBus
2021-07-17 11:34:03 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 1661
2021-07-17 11:34:03 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-07-17 11:34:03 | razer                          | INFO     | Found device.0: 0003:1532:0209.0004
2021-07-17 11:34:03 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowChromaTournamentEdition
2021-07-17 11:34:03 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2021-07-17 11:34:03 | razer                          | INFO     | Serving DBus
2021-07-17 11:40:12 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 142
2021-07-17 11:40:13 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 389, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 142
2021-07-17 12:00:33 | razer.device0.keymanager       | INFO     | Got game mode combo
2021-07-17 12:00:34 | razer.device0.keymanager       | INFO     | Got game mode combo
2021-07-17 12:01:24 | razer.device0.keymanager       | INFO     | Got macro combo
2021-07-17 12:28:31 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 1590
2021-07-17 12:28:31 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-07-17 12:28:31 | razer                          | INFO     | Found device.0: 0003:1532:0209.0004
2021-07-17 12:28:31 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowChromaTournamentEdition
2021-07-17 12:28:31 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2021-07-17 12:28:31 | razer                          | INFO     | Serving DBus
2021-07-17 12:29:08 | razer.device0.keymanager       | INFO     | Got macro combo
2021-07-17 12:29:25 | razer                          | INFO     | Stopping daemon.
2021-07-17 12:29:27 | razer                          | INFO     | Initialising Daemon (v3.0.1). Pid: 2789
2021-07-17 12:29:27 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-07-17 12:29:27 | razer                          | INFO     | Found device.0: 0003:1532:0209.0004
2021-07-17 12:29:27 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowChromaTournamentEdition
2021-07-17 12:29:27 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2021-07-17 12:29:27 | razer                          | INFO     | Serving DBus
2021-07-17 13:08:37 | razer.device0                  | INFO     | Suspending RazerBlackWidowChromaTournamentEdition
2021-07-17 13:10:22 | razer                          | WARNING  | Removing 0003:1532:0209.0004
2021-07-17 13:10:45 | razer                          | INFO     | Found valid device.0: 0003:1532:0209.000B
2021-07-17 13:10:45 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowChromaTournamentEdition
2021-07-17 13:10:45 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2021-07-17 13:26:26 | razer.device0                  | INFO     | Resuming RazerBlackWidowChromaTournamentEdition
2021-07-17 13:28:32 | razer.device0.keymanager       | INFO     | Got macro combo
2021-07-17 13:28:42 | razer                          | INFO     | Stopping daemon.
```

z3ntu, 2021-07-25:
Try adding 142 to https://github.com/openrazer/openrazer/blob/5ddf10eb85a803c83fa31465c1df4bdf0e860a17/daemon/openrazer_daemon/keyboard.py#L119 . This file should exist in `/usr/lib/python3.9/site-packages/openrazer_daemon/keyboard.py` or `/usr/lib/python3/dist-packages/openrazer_daemon/keyboard.py` or somewhere like this on your filesystem - you can edit it there directly and restart the daemon.

adityabisoi, 2021-07-26:
I tried adding `142:'MACROMODE`. It seems the previous error in log is not showing anymore, but the daemon still stops, with the following in log as last line
```
 2021-07-26 21:19:03 | razer.device0.keymanager       | INFO     | Got macro combo
```

adityabisoi, 2021-07-26:
Is there a way to log the key IDs that I press? Maybe I can fix this if I know the ID of each key.

adityabisoi, 2021-07-28:
@z3ntu Could you please take a look at this?

z3ntu, 2021-08-01:
Ok I think that the exception wasn't the cause of the error then but I also don't really have another idea either...

closed 2022-06-19 by github-actions[bot]